webcamoid/webcamoid #100: Crash in MultiSrcElement::codecLibUpdated

yurivict, 2017-12-01:
It crashes during startup in the function ```MultiSrcElement::codecLibUpdated``` in this line:
```libAvKys/Plugins/MultiSrc/src/multisrcelement.cpp:204: auto media = this->m_mediaSource->media();```

Found on FreeBSD 11.1 while trying to create a port.

yurivict, 2017-12-01:
This looks like there might be some memory corruption.
Did you run webcamoid with google perftools library checking memory operations?
Did you run it with valgrind's memcheck tool?

hipersayanX, 2017-12-01:
I use Valgrind in GNU/Linux and Mac, and Dr Memory in Windows, and non of them mark an error there. Will check google perftools.
Not sure if it will work but, try changing from static_cast to reinterpret_cast [here](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/MultiSrc/src/multisrcelement.cpp#L31).

yurivict, 2017-12-01:
changing from static_cast to reinterpret_cast didn't help. -(

hipersayanX, 2017-12-01:
What if you put a qDebug before this->m_mediaSource->media().

    qDebug() << this->m_mediaSource;
    this->m_mediaSource->media();

or 

    qDebug() << this->m_mediaSource.data();
    this->m_mediaSource->media();

does it print a valid direction (!= 0x0)?

yurivict, 2017-12-01:
It prints:
```
this->m_mediaSource= QSharedPointer(MediaSource(0x819e40400))
this->m_mediaSource.data()= MediaSource(0x819e40400)
```

hipersayanX, 2017-12-01:
I quite lost here, this->m_mediaSource is a valid pointer contained inside a QSharedPointer, even so it throw an exception there. The only thing I can think is that QSharedPointer is freeing the pointer but not clearing it (which is not the normal behavior of QSharedPointer). Let's try an experiment, keep the previous qDebug, and replace ptr_init with:

    template<typename T>
    inline QSharedPointer<T> ptr_init(QObject *obj=nullptr)
    {
        if (!obj)
            return QSharedPointer<T>(new T(), [](T *ptr) {
                qDebug() << "Deleting (Base) " << ptr;
            });
    
        return QSharedPointer<T>(static_cast<T *>(obj), [](T *ptr) {
            qDebug() << "Deleting " << ptr;
        });
    }

It may print something like "Deleting MediaSource(0x819e40400)" before _qDebug() << this->m_mediaSource.data()_, if it's deleting the pointer.

yurivict, 2017-12-01:
No, it didn't print anything extra.

hipersayanX, 2017-12-01:
Try, replacing

    auto media = this->m_mediaSource->media();
    auto loop = this->m_mediaSource->loop();
    auto showLog = this->m_mediaSource->showLog();

with

    QString media;
    bool loop = true;
    bool showLog = true;

yurivict, 2017-12-02:
With
```
QString media;
bool loop = true;
bool showLog = true;
```
it starts and shows video. No menus are displayed on the sides, it opens only with video and blank on the left and on the right.

hipersayanX, 2017-12-02:
Well, at least it launched. Does the options in Preferences appears?

yurivict, 2017-12-02:
They appear when clicked on the menu icon.

hipersayanX, 2017-12-02:
Good! what it say in General Options? are all options filled or is something missing?

yurivict, 2017-12-02:
It looks like this:
[image]

hipersayanX, 2017-12-02:
Everything seems correct, so I will return to this:

>No menus are displayed on the sides, it opens only with video and blank on the left and on the right.

When you press the webcam icon, everything is blank, no green button, no message showing webcam no found, no options, right?

[image]

yurivict, 2017-12-02:
There is the green button, it looks similar, except background is lighter gray.
So, I guess, it works fine. Just that line was removed.

hipersayanX, 2017-12-02:
Then the problem is solved, the color depends on desktop theme. Just remove the previous qDebugs and restore the original ptr_init. I will keep this issue open to test further, still not sure whats wrong with the pointer.

yurivict, 2017-12-02:
Ok. I will add the link to this bug into the patch.
Thanks!

yurivict, 2017-12-04:
It still crashes. Something changed, like some dependency got updated, and it crashes again.

Now it passes the point in MultiSrcElement::codecLibUpdated that was patched, and crashes inside
>    this->m_mediaSource =
>            ptr_init<MediaSource>(this->loadSubModule("MultiSrc", codecLib));

while deleting the previous this->m_mediaSource.

The interesting thing is that the constructor ```MultiSrcElement::MultiSrcElement``` is never called. Print statement there that I put never prints anything. So, invocation the method ```MultiSrcElement::codecLibUpdated``` is invalid because the object hasn't been created.

I think, you might be creating the wrong type and casting?

yurivict, 2017-12-04:
I've put printouts in the beginning and end of ```MediaSource::MediaSource```. The object ``MediaSource(0x80aad7d20)``` is being created, and from inside the constructor the never-created object ```MediaSource(0x818a47410)``` is being destroyed. This is obciously the cause: you destroy the un-created object ```MediaSource(0x818a47410)```.

```
>>> MediaSource::MediaSource this= MediaSource(0x80aad7d20)
MultiSrcElement::MultiSrcElement: m_mediaSource= QSharedPointer(MediaSource(0x818a47410))
after crash #1 before diving into crash#2: this->m_mediaSource= QSharedPointer(MediaSource(0x818a47410))
<<< MediaSource::~MediaSource this= MediaSource(0x818a47410)
```

hipersayanX, 2017-12-05:
Found the problem. Is a bug in clang 4.0.0, that is the version in FreeBSD repositories, that's the reason because this error doesn't appears in GNU/Linux and Mac, they use clang >= 5.0.0. Switching to GCC in FreeBSD works perfectly well. My guess is because the base class (MediaSource in this case) has all it's methods virtual, this is allowed in C++ (the class in not pure virtual) but, clang is taking it as it were.

**Note aside**: Not sure if it's a problem with VirtualBox but, Qt Quick Controls 2 based apps doen't works well in FreeBSD/TrueOS, it doesn't not show any text even with software renderer enabled, if that the case please report the bug to FreeBSD, TrueOS and Qt bug tracker.

yurivict, 2017-12-05:
> Found the problem. Is a bug in clang 4.0.0

Do you have the bug report number? Because clang is actively maintained, and they fix all such bugs.

yurivict, 2017-12-05:
It crashes when built with clang-5.0.0 too.

hipersayanX, 2017-12-05:
Did you tried with GCC?

yurivict, 2017-12-05:
I am building with gcc6 now.

yurivict, 2017-12-05:
gcc6 for some reason didn't build these files:
```
Error: Missing: lib/avkys/submodules/ACapsConvert/libffmpegav.so
Error: Missing: lib/avkys/submodules/ACapsConvert/libffmpegsw.so
Error: Missing: lib/avkys/submodules/AudioDevice/libalsa.so
Error: Missing: lib/avkys/submodules/AudioDevice/libjack.so
Error: Missing: lib/avkys/submodules/AudioDevice/liboss.so
Error: Missing: lib/avkys/submodules/AudioDevice/libpulseaudio.so
Error: Missing: lib/avkys/submodules/AudioDevice/libqtaudio.so
Error: Missing: lib/avkys/submodules/MultiSink/libffmpeg.so
Error: Missing: lib/avkys/submodules/MultiSink/libgstreamer.so
Error: Missing: lib/avkys/submodules/MultiSrc/libffmpeg.so
Error: Missing: lib/avkys/submodules/MultiSrc/libgstreamer.so
Error: Missing: lib/avkys/submodules/VideoCapture/libffmpeg.so
Error: Missing: lib/avkys/submodules/VideoCapture/libgstreamer.so
Error: Missing: lib/avkys/submodules/VideoCapture/libv4l2sys.so
Error: Missing: lib/avkys/submodules/VideoCapture/libv4lutils.so
Error: Missing: lib/avkys/submodules/VirtualCamera/libffmpeg.so
Error: Missing: lib/avkys/submodules/VirtualCamera/libgstreamer.so
Error: Missing: lib/avkys/submodules/VirtualCamera/libv4l2sys.so
Error: Missing: lib/avkys/submodules/VirtualCamera/libv4lutils.so
```
They are missing compared to clang build.

hipersayanX, 2017-12-05:
Does it show this table at beginning?

```
Checking for alsa... yes
Checking for avfoundation... no
Checking for cmio... no
Checking for coreaudio... no
Checking for dshow... no
Checking for ffmpeg... yes
Checking for ffmpeg_avcodec_contextframerate... yes
Checking for ffmpeg_avcodec_extracodecformats... yes
Checking for ffmpeg_avcodec_freecontext... yes
Checking for ffmpeg_avcodec_packetref... yes
Checking for ffmpeg_avcodec_rescalets... yes
Checking for ffmpeg_avcodec_sendrecv... yes
Checking for ffmpeg_avcodec_subtitledata... yes
Checking for ffmpeg_avformat_allocoutputcontext... yes
Checking for ffmpeg_avformat_codecpar... yes
Checking for ffmpeg_avutil_extraoptions... yes
Checking for ffmpeg_avutil_extrapixformats... yes
Checking for ffmpeg_avutil_framealloc... yes
Checking for ffmpeg_avutil_sampleformat64... yes
Checking for ffmpeg_avresample... yes
Checking for ffmpeg_swresample... yes
Checking for gstreamer... yes
Checking for jack... yes
Checking for libuvc... no
Checking for oss... yes
Checking for pulseaudio... yes
Checking for qtaudio... yes
Checking for v4l2... yes
Checking for v4l2_extendedcontrols... yes
Checking for v4lutils... yes
Checking for vcamwin... no
Checking for wasapi... no
```

yurivict, 2017-12-05:
With gcc6 it can't find ffmpeg. With clang it was found.

hipersayanX, 2017-12-06:
Look at libAvKys/config.log, there must be a section called "executing config test ffmpeg", there must be an explanation of the error bellow there.
Also be sure to work with a clean tree after switching to GCC.

yurivict, 2017-12-06:
It's looking for g++, but my compiler is called g++6. I don't know why.

yurivict, 2017-12-06:
It fills a lot of CXX and LINK variables in Makefiles with g++ instead. There is some bug, but I can't find it. Maybe you strip digits from g++N somewhere?

yurivict, 2017-12-06:
As a temporary workaround, I created the symlink g++ -> g++6.
With this symlink it still fails:
```g++: error: unrecognized command line option '-framework'```

hipersayanX, 2017-12-06:
You can configure the compiler with:

```
qmake -spec freebsd-g++ Webcamoid.pro QMAKE_CXX=/usr/bin/g++6
```

hipersayanX, 2017-12-06:
>g++: error: unrecognized command line option '-framework'

Also, don't get confused by this kind of messages, Qmake compile all test cases to see what is available and what is not, including tests from other systems.
There must be something related to -lavformat -lavcodec -lswscale -lavutil  somewhere there.

yurivict, 2017-12-06:
It works with g++6. But I can't update the port because of this g++ problem. It might be a bug in qmake that it accidentally overrides QMAKE_CXX somewhere.

hipersayanX, 2017-12-06:
Just in case, my test environment:

TrueOS stable
clang 4.0.0
gcc 5.4.0
Qt 5.7.1
Qt Creator 4.2.2

hipersayanX, 2017-12-12:
I was thinking, probably this bug is caused by in stack exhaustion somewhere in the code, I remember had a similar problem in the past, with code crashing were is not supposed to crash, very hard to debug, hope gperftools can give me a clue there.

Any way I'm more worried about this:

>Note aside: Not sure if it's a problem with VirtualBox but, Qt Quick Controls 2 based apps doesn't works well in FreeBSD/TrueOS, it doesn't not show any text even with software renderer enabled, if that the case please report the bug to FreeBSD, TrueOS and Qt bug tracker.

I've tested a very basic QQC2 qml, and get the same problem, it must be reported and fixed as quick as possible.

yurivict, 2017-12-12:
> I was thinking, probably this bug is caused by in stack exhaustion somewhere in the code

No, this isn't a stack exhaustion problem. The stack is only 19-deep.

This is a show-stopper for clang-based systems.

hipersayanX, 2017-12-12:
>This is a show-stopper for clang-based systems.

But it works well in Mac and GNU/Linux, both with clang.

yurivict, 2017-12-12:
Memory might be arranged differently.

yurivict, 2018-01-25:
If you are could create a small testcase demonstrating that this is a bug in C++, I would be happy to report it to clang.

So far, I am not convinced that this is a bug in clang.

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

closed 2019-01-22 by hipersayanX