webcamoid/webcamoid #101: The application window does not appear

vporton, 2017-12-02:
Debian webcamoid 8.1.0+dfsg-2

When I start webcamoid, no windows are opened, so I do not see the interface of the program.

May be related with https://bugzilla.gnome.org/show_bug.cgi?id=791122

$ webcamoid 
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:25 module "AkQml" is not installed

hipersayanX, 2017-12-02:
Di you installed from repositories? If you did then the required Qml files must be located in [/usr/lib/x86_64-linux-gnu/qt5/qml/AkQml](https://packages.debian.org/sid/amd64/akqml/filelist)

>May be related with https://bugzilla.gnome.org/show_bug.cgi?id=791122

No, Webcamoid has its own effect framework independent from the system.

vporton, 2017-12-02:
> 5493e4bb","name":"GitHub"},"entity":{"external_key":"github/webcamoid
> /webcamoid","title":"webcamoid/webcamoid","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/ass
> ets/143418/17495839/a5054eac-5d88-11e6-95fc-
> 7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent
> .com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
> b52498112777.png","action":{"name":"Open in GitHub","url":"https://gi
> thub.com/webcamoid/webcamoid"}},"updates":{"snippets":[{"icon":"PERSO
> N","message":"@hipersayanX in #101: Di you installed from
> org/sid/amd64/akqml/filelist)\r\n\r\n\u003eMay be related with https:
> //bugzilla.gnome.org/show_bug.cgi?id=791122\r\n\r\nNo, Webcamoid has
> system.\r\n"}],"action":{"name":"View Issue","url":"https://github.co
> m/webcamoid/webcamoid/issues/101#issuecomment-348708285"}}}

Yes, I installed from official Debian repositories.

I have /usr/lib/x86_64-linux-gnu/qt/qml/AkQml (without "5" after "qt").

hipersayanX, 2017-12-02:
Try launching Webcamoid with:

    webcamoid -q /usr/lib/x86_64-linux-gnu/qt/qml

vporton, 2017-12-02:
@hipersayanX It worked.

Weirdly after this it now works even if I start it without `-q /usr/lib/x86_64-linux-gnu/qt/qml`.

But:

```
$ webcamoid -q /usr/lib/x86_64-linux-gnu/qt/qml
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model"
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/Control.qml:90: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/ComboBox.qml:397: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/ComboBox.qml:453: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/TextInputWithHandles.qml:104: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/TextInputWithHandles.qml:131: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/EditMenu_base.qml:46: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/Control.qml:90: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/BasicButton.qml:194: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/Control.qml:90: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/BasicButton.qml:194: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/Control.qml:90: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/BasicButton.qml:194: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/Control.qml:90: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/BasicButton.qml:194: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/Control.qml:90: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/TextField.qml:656: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/TextInputWithHandles.qml:104: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/TextInputWithHandles.qml:131: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/EditMenu_base.qml:46: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/Control.qml:90: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/TextField.qml:656: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/TextInputWithHandles.qml:104: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/TextInputWithHandles.qml:131: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/EditMenu_base.qml:46: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/Control.qml:90: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/BasicButton.qml:194: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/Control.qml:90: ReferenceError: parent is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Private/BasicButton.qml:194: ReferenceError: parent is not defined
```

hipersayanX, 2017-12-02:
Once you pass the -q option it keep the path registered in _~/.config/Webcamoid/Webcamoid.conf_, so you don't need it again.
The other messages are external to Webcamoid, it may be a problem in your install or something, you must report that to Debian bug tracker.

hipersayanX, 2017-12-03:
Also please, it will be very helpful if you can give an advice to Debian packagers to use [INSTALLQMLDIR](https://github.com/webcamoid/webcamoid/wiki/Raw-build-and-install#install-paths) when building the package, that may be the root problem here.

hpfn, 2017-12-04:
Hi,

The use of INSTALLQMLDIR is an easy fix. Thanks for the solution!

But let me ask a question: Webcamoid 8.1.0 changed the path by [libAvKys/commons.pri](https://anonscm.debian.org/cgit/collab-maint/webcamoid.git/diff/libAvKys/commons.pri?id=0fe63c1dafeda60d171285b06f12246d7f6b996d).

Is it better to go back to the default 'qt5' path? 

Maybe libAvKys/AkQml/AkQml.pro needs to be edited too. (I am really do not know)

Regards,
Herbert

hipersayanX, 2017-12-04:
@hpfn but there is not standard or default path, each distro do what they want, as example the Qml install for each distro:

Arch: /usr/lib/qt/qml
Debian/Ubuntu: /usr/lib/x86_64-linux-gnu/qt5/qml
Fedora: /usr/lib64/qt5/qml
OpenSUSE:  /usr/lib64/qt5/qml

And this is just for x86_64 architecture. And looking and Mac an Windows, doesn't make it much better:

Mac: /usr/local/Cellar/qt/5.9.3/qml
Windows: C:/Qt/5.9/mingw53_32/qml

And I'm still missing the paths for Qt official binary distributions, MSYS2 and others.
At some point you may think it can be solved using $$[QT_INSTALL_QML] as suggested [here](https://github.com/webcamoid/webcamoid/pull/103), but then it break install and deploy scripts in Mac, Windows and other systems.
The only way to deal with this fragmented ecosystem was creating a variable controlling that path.

hpfn, 2017-12-04:
Em 04-12-2017 14:20, hipersayanX escreveu:
> @hpfn <https://github.com/hpfn> but there is not standard or default path, each distro do what they want, as example the Qml install for each distro:
>
> Arch: /usr/lib/qt/qml
> Debian/Ubuntu: [...]

Ok. I only know Debian.

Thanks for detailed explanation!

yurivict, 2017-12-06:
Window doesn't appear on one of my computers too. Same system - FreeBSD 11, same package, binary-identical. Webcam otherwise works fine there.

hpfn, 2017-12-06:
"same system"

There is no Webcamoid available on Debian KFreeBSD-*.

yurivict, 2017-12-06:
I run on FreeBSD. It is available there in ports.

QBoulevard, 2018-02-10:
I am using Linux Rebecca. Application window appears then disappears and is no longer open. I have tried deleting the folder, then reinstalling multiple times and also tried launching through the terminal. I saw I needed to install plugins, but when I checked, I saw I already had these plugins. The notices in my terminal said reinstalling Webcamoid might help. I have tried six or seven reinstalls now but the exact same thing happens. It opens momentarily, then disappears and is no longer running. It is too bad. It seems like a very good program and much more flexible than Cheese.

hipersayanX, 2018-02-10:
@QBoulevard what is the output message from terminal? Can you upload the [config files](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#gnulinux) to [gist](https://gist.github.com/)?

QBoulevard, 2018-02-10:
Unfortunately I have restarted several times since I had the terminal open. It mentioned a few plugins but I wasn't sure if these were plugins specific to the application or some general type. I am wondering if the problem is something to do with my webcam itself. Seems to work in Cheese though. I want to be able to use a better quality microphone, not the one supplied with the webcam. I am trying to record a speech I am giving live in front of an audience, and ideally I would wear the microphone, but with Linux I cannot use the mic jack so I am limited to USB mic. I can only use the webcam mic with Cheese, but I can record separately with Webcamoid.

hipersayanX, 2018-02-10:
@QBoulevard I need that logs to tell if that is a problem related to Webcamoid or not, there is not other way i can debug it.

QBoulevard, 2018-02-11:
It's okay. I think it will be better if I record the speech with my new iphone. I am traveling to the occasion by bus so phone will be much easier to carry than a laptop, and less conspicuous.

QBoulevard, 2018-02-23:
Hi. I reinstalled Linux, and now the version I have is Serena. I reinstalled for another reason. I tried installing Webcamoid and the exact same thing happened. I was delighted because I really want to do a YouTube without looking like a total fool. Cheese is useless (sound quality sucks). So I was so hopeful that Webcamoid would work for me. Unfortunately, after I restarted just now, Webcamoid is doing the exact same thing. It starts but then, disappears. Now in the terminal it says, "aborted after line 20: 3202."
Can anyone help me out with this? I have toyed with the thought that it is "disappearing" to another workspace but honestly I cannot find it anywhere. I believe it crashes instantly, within a fraction of a second after it starts. I believe this has to do with restarting the computer. This is the same thing that happened last time.

QBoulevard, 2018-02-23:
So that is it, "Line 20: 3202," where it crashes. Until then it is loading.

QBoulevard, 2018-02-23:
And the program works before I restart. After I restart I cannot use Webcamoid at all.

hipersayanX, 2018-02-24:
@QBoulevard, I don't want to be rude but I don't want to read your life story, for the third and last time, **just give me the terminal output log**.

QBoulevard, 2018-02-24:
Sorry, you are indeed being rude. I don't even know what terminal input log is. I got Webcamoid working on my own and the sound quality sucks very badly, so I am not using it. If you are rude, other will not want your "help."

hipersayanX, 2018-07-05:
[wiki](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-GNU-Linux) updated, there is the information on required dependencies and launch options.

closed 2018-07-05 by hipersayanX

adatum, 2019-02-18:
On Fedora 29, I had the same issue as OP and had to use `webcamoid -q /usr/lib64/qt/qml/` to launch the first time.