webcamoid/webcamoid #102: Files installed into lib/qt/qml instead of lib/qt5/qml

yurivict, 2017-12-03:
It installs these two files:
```
lib/qt/qml/AkQml/libAkQml.so
lib/qt/qml/AkQml/qmldir
```
Your rpm spec ```ports/rpm/webcamoid.spec``` says that they should be in ```%{_libdir}/qt5/qml/AkQml```, but your makefile installs them into ```lib/qt/qml```.

yurivict, 2017-12-03:
https://github.com/webcamoid/webcamoid/pull/103

hipersayanX, 2017-12-03:
Package build scripts has not been updated since long time ago, I was waiting for more distros to package Webcamoid, and it were intended to be removed before 9.x series.
Nowadays Webcamoid is packaged in [many distros](https://repology.org/metapackage/webcamoid), and I have automated builds for developers and packagers, and binary packages for users, so I don't need them anymore, will remove soon.

closed 2017-12-04 by yurivict