webcamoid/webcamoid #104: The ASCII art plugin uses too large letters even with the smallest font size

yurivict, 2017-12-21:
The smallest font size I can set is 6, but with 6 it is still to coarse. Letters with size 6 appear much larger than if they would have been normally displayed on the screen. I think, you scale them.

IMO, you should add scaling factor as a separate combobox or spin-control. This way the user will have more control.

hipersayanX, 2017-12-22:
Charify and Matrix plugins recreate the original frame preserving (more or less) the original resolution, if the a character has a size of 15x20 in screen, it will have a size of 15x20 in the frame when you take a photo, record a video, or send the frame to the virtual camera.
If you shrink or maximize the window, the characters will appear smaller or larger because of frame scaling.

hipersayanX, 2018-12-27:
Font point size can be changed to lower values in font selection dialog, adding a new slider to control that is redundant.

closed 2018-12-27 by hipersayanX