webcamoid/webcamoid #106: MP4 Crashes Windows 10 64-Bit

ellisgl, 2017-12-26:
### Summary
Recording with MPEG4 with audio on = crash.
With audio turned off, there is 0 byte saved. Also noticed, there is no audio in any other codec.

### Expected Behavior
Not to crash and record audio.

### Your Environment
* Webcamoid information: 8.1.0 64 Bit
* Operating System information: Windows 10 Pro 64-bit
* Any other useful information: 
Faulting application name: webcamoid.exe, version: 8.1.0.0, time stamp: 0x00000000
Faulting module name: avformat-57.dll, version: 57.56.101.0, time stamp: 0x00000000
Exception code: 0xc0000005
Fault offset: 0x00000000000c9de6
Faulting process id: 0x478c
Faulting application start time: 0x01d37e88a2694894
Faulting application path: C:\Program Files\webcamoid\bin\webcamoid.exe
Faulting module path: C:\Program Files\webcamoid\bin\avformat-57.dll
Report Id: 4cafd6c6-1477-4268-ba81-da0f1aca22c5
Faulting package full name: 
Faulting package-relative application ID:

hipersayanX, 2017-12-27:
Thanks! will check this.

ellisgl, 2017-12-27:
hipersayanX - Strange thing was that it worked last night for 5 minutes then I started to change which codec, then it stopped working. Uninstalled, installed the 32 bit version and it still happened (minus working for the 5 minutes). Uninstalled, rebooted, reinstalled 64 bit version and decided to report it, since I don't want to use BandiCam. =)

hipersayanX, 2017-12-27:
You can try [resetting all configs](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#windows).

ellisgl, 2017-12-27:
OK - uninstalled, removed the registry key, reinstalled. I can record MPEG4 (MJPEG) video without crashing, but still no sound. I tried two sound sources, one from a USB sound adapter and one from DroidCam audio.  Both audio source record fine in audacity.

hipersayanX, 2018-12-29:
I think this is actually #93, will close this since there are more info there.

closed 2018-12-29 by hipersayanX