webcamoid/webcamoid #107: Interface full of glitches on Mac High Sierra 10.13.2

ondraspa, 2018-01-16:
### Summary

The interface does not show any icons and text is full glitches.

* Webcamoid information: (8.1.0) (Mac)
* Operating System information: (Mac High Sierra) (10.13.2) (unix)
* Any other useful information:  screenshots
[image]

hipersayanX, 2018-01-16:
What graphics card are you using?

ondraspa, 2018-01-16:
Intel integrated 4600

hipersayanX, 2018-01-16:
Try running webcamoid as:

QT_QUICK_BACKEND=software /Applications/webcamoid.app/Contents/MacOS/webcamoid

ondraspa, 2018-01-16:
It works.

hipersayanX, 2018-01-16:
Great! it will be fixed in 8.5.0.

ondraspa, 2018-01-16:
OK thanks a lot.

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

closed 2019-01-22 by hipersayanX