webcamoid/webcamoid #108: ffmpeg 3.5

hpfn, 2018-01-25:
# Reporting a problem?

ffmpeg 3.5 release

### Summary

I received a report[0] that describe a FTBFS with the upcoming version
3.5 of ffmpeg. Thanks Jcowgill from Debian. The report maybe help.
Build log[1].

[0] - https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=888371

"
While 3.5 has not been finalized and the ABI is not stable yet, there should
not be any significant API breakages before the release.

Incomplete list of changes (based on looking at common build failures):
- Some fields in AVCodecContext have been removed and replaced with private
  options which can be set using the av_opt_set* APIs
- Most CODEC_* constants have been renamed to AV_CODEC_*
- The buffer constants FF_INPUT_BUFFER_PADDING_SIZE and FF_MIN_BUFFER_SIZE
  have been renamed to AV_INPUT_BUFFER_PADDING_SIZE and
  AV_INPUT_BUFFER_MIN_SIZE.
- The old resampling API provided by libavcodec has been removed. Use
  libswresample instead.
- The libavfilter/avfiltergraph.h header has been removed, include
  libavfilter/avfilter.h instead.
- The AVFrac structure (representing mixed rational numbers) has been
  removed.

Build log:
[1] - https://people.debian.org/~jcowgill/ffmpeg-3.5-20180122/webcamoid_amd64.build
"

Regards,
Herbert

hpfn, 2018-07-20:
Hi,

Webcamoid was set to autoremove on Debian. 

There is a plan to release a new version with the fix? Or it is better to apply a patch?

Regards,

ROBERT-MCDOWELL, 2018-07-20:
maybe you can solve the problem yourself and contribute a patch to the community?

hpfn, 2018-07-20:
I do the Debian package so I can apply the patch if no release is planned.
That's all.

The issue was closed on May 1. I do not need to solve the problem myself.

I study Python BTW.

hipersayanX, 2018-07-21:
A patch was already [applied](https://github.com/webcamoid/webcamoid/pull/119) in development tree, can you apply those patchs to your packages?
I'm working in Linux virtual camera driver right now and it will take more or less 2 months maybe, a new release will come right after finishing with that, maybe 3 months in total with bug fix.

hpfn, 2018-07-21:
Em 21-07-2018 00:58, hipersayanX escreveu:
> A patch was already applied <https://github.com/webcamoid/webcamoid/pull/119> in development tree, can you apply those patchs to your packages?
> I'm working in Linux virtual camera driver right now [...]

I will try to do the package now.

THe package has this RC bug. But I saw the Debian release plan.
No problem about time.

[2019-Jan-12] Transition freeze
[2019-Feb-12] Soft freeze (no new packages, no re-entry, 10-day migrations)
[2019-Mar-12] Full freeze

hipersayanX, 2018-07-21:
Syncing with distro releases right now is a bit hard that, if we can't reach a new release before freeze, users still has AppImage, the portable and the installer as an alternative, or they can continue using Debian Testing.

hipersayanX, 2018-07-22:
No problem, the extern "C" {} may only surround the #includes, everything else is ok.

candrews, 2018-11-05:
Since this issue has been fixed with the merging of https://github.com/webcamoid/webcamoid/pull/119 can this issue please be closed?

hipersayanX, 2018-11-06:
When an issue is marked in **Critical** category it must be closed after release, because it was fixed in mainline but not in the version owned by the user.
When an issue is marked in **Veredict** category it can be closed right away, when user solve it's question, or after release if user did not provided any feedback, ultimately the user can close it issue whenever it want.

PR #119 (2018-05-01): Unbreak build against FFmpeg 4.0
Fixes #108. I've only tested build: no webcam here.