webcamoid/webcamoid #110: "Build and install the RPM package" wiki page instructions break on rpmbuild

giacecco, 2018-02-16:
### Summary

"Build and install the RPM package" wiki page instructions at https://github.com/webcamoid/webcamoid/wiki/Build-and-install-the-RPM-package break on rpmbuild step, running on Fedora 27 and rpmbuild 4.14.1-1.fc27

### Current Behavior

When doing `rpmbuild -ba ~/webcamoid.spec` I get 

```
error: parse error in expression
error: /home/giacecco/webcamoid.spec:131: bad %if condition
```

### Expected Behavior

I don't know, I had never done this before.

### Steps to Reproduce

Follow the instructions by the letter.

Thanks!

hipersayanX, 2018-02-16:
Try removing all lines from [119 to 131 and line 135](https://github.com/webcamoid/webcamoid/blob/master/ports/rpm/webcamoid.spec#L119), and leave the qmake part.

hipersayanX, 2018-07-05:
Distro specific build scripts were removed from tree in favor of [installer and portable build](https://github.com/webcamoid/webcamoid/wiki/Deploy-and-create-self-contained-binary-packages) scripts. Distributions developers wanting to package Webcamoid for their distros can use [deploy scripts](https://github.com/webcamoid/webcamoid/blob/master/ports/ci/travis/install_deps.sh#L59) and other distributions [build scripts](https://repology.org/metapackage/webcamoid/versions) as a guide.

closed 2018-07-05 by hipersayanX