webcamoid/webcamoid #111: windows build configuration / vs version + qt version

soliton4, 2018-02-20:
i searched the issues, the wiki and parts of the sources, but i didnt find a clear answer to what the exact windows build configuration is.

### Summary

i am using qt 5.9.4 with visual studio 2013 64bit
i get some minor compiler that are easily fixed but it lets me suspect that vs 2013 might be to old.
can you give me your detailed description of your windows build environment?

- qt version
- visual stuido or mingw
- if visual studio, which version?

### Current Behavior

a debug build finished after removing 
TRANSLATIONS_PRI = ../translations.pri
from the .pro file
the program doesnt seem to run properly thou. i waited several minutes for it to start up.

the release build has several issues

first issue is this:
C:\Users\paperspace\projects\build-Webcamoid-Desktop_Qt_5_9_4_MSVC2013_64bit-Release\StandAlone\.qtquickcompiler\qtquickcompiler_loader.cpp:22: error: C1083: Cannot open include file: 'private/qv4function_p.h': No such file or directory

### Expected Behavior

i am under the impression my build setup is not correct. i expect the build to run through and to be able to run a release build and debug a debug build.

should i upgrade to vs2015? vs2017? or mingw?
should i use qt 5.7.0 instead of 5.9.4?

### Steps to Reproduce

install vs2013 update 5
install qt 5.9.4
configure project according to wiki and as visual studio 2013 64bit

### Suggestions and tips

it might be helpful to have that information in the wiki. or maybe i just didnt find it so maybe a more prominent mention of the detailed build setup in the build windows section of the wiki

### Your Environment

* Webcamoid information: latest version from github as of today (2018-02-20)
* Operating System information: windows server 2016 (basically windows 10) (64 bit intel)

ps: great project ;)

hipersayanX, 2018-02-21:
> can you give me your detailed description of your windows build environment?

>qt version

Qt 5.10 or the latest one in existence.

>visual stuido or mingw

Normally MinGW, Visual Studio for 64 bits (under Windows), and the release version is always build with MinGW under GNU/Linux.
Always under Qt Creator.

>if visual studio, which version?

I must update the wiki, but VS is not required any more, you only need the [**Build Tools for Visual Studio**](https://www.visualstudio.com/downloads/).
According to AppVeyor it must be possible to build the project with [MSVC 2013 and higher](https://ci.appveyor.com/project/hipersayanX/webcamoid), but in my case only MSVC 2017 Build Tools worked for me, and Qt Creator did not recognized CDB at all :confused: 

> a debug build finished after removing
> TRANSLATIONS_PRI = ../translations.pri
> from the .pro file
> the program doesnt seem to run properly thou. i waited several minutes for it to start up.
>
> the [...]

Are you building from VS or Qt Creator? if building from Qt Creator, you must set **Default build directory** to "." (dot).

>should i upgrade to vs2015? vs2017? or mingw?

As I said before, it must be able to build with MSVC2013 and higher, and MinGW build is always guaranted. If you want to build Webcamoid for 64 bits with MinGW you can install [MSYS2](http://www.msys2.org/) (since official Qt doesn't provides 64 bit MinGW builds).

>should i use qt 5.7.0 instead of 5.9.4?

5.9 is ok, and it will be the minimum requirement soon.

>it might be helpful to have that information in the wiki. or maybe i just didnt find it so maybe a more prominent mention of the detailed build setup in the build windows section of the wiki

I don't know much of VS, but if you can make it work there consider writting an entry in the [wiki](https://github.com/webcamoid/webcamoid/wiki) :smile:

>ps: great project ;)

Thanks!

soliton4, 2018-02-22:
thank you for your response.
i set up a new fresh windows server 2016 for a build attempt
i installed 
- visual studio 2017
- qt 5.9.4

ffmpeg:  
[image]

qt config:  
[image]

project config:  
[image]

project config detailed:
[image]

additional arguments:  
FFMPEGINCLUDES=C:\Users\paperspace\projects\ffmpeg\include FFMPEGLIBS=-LC:\Users\paperspace\projects\ffmpeg\lib FFMPEGLIBS+=-lavformat FFMPEGLIBS+=-lavcodec FFMPEGLIBS+=-lswresample FFMPEGLIBS+=-lswscale FFMPEGLIBS+=-lavutil FFMPEGLIBS+=-lavdevice

ffmpeg files copied:
[image]

run arguments:  
[image]

compile output (sorry using gist for the first time):  
https://gist.github.com/soliton4/a285f61931a1a37a4708c2b4440cecc6

issues:  
[image]

hipersayanX, 2018-02-23:
Get it, in the second capture, select MSVC and MinGW targets, UWP is not supported yet, and anyway the virtual camera will not work there because UWP apps are sandboxed.

In the fourth capture deselect shadow build, and in _Tools > Options > Build & Run > Default Build Directory_ write a **"."** (dot). Again In the fourth capture where it say _Build & Run > Run_ (in the left of your capture), select that and go to the bottom where it say **Run Environment** in **PATH** field add the path where your FFmpeg's DLL are located, so you will not need to copy those files again when you clean the directory.

Everything else seems to be ok.

soliton4, 2018-02-23:
thanks for coming back to me
i promise i will make a wiki entry on how to build on windows but first i have to figure out how

i followed your steps, deleted my pro.user file and this is my new config:
[image]

i did a build and got this error:
[image]

i figured i should do a clean. running clean deletes some of the files in the repo so i had to do a checkout . again

this is my new compile output
https://gist.github.com/soliton4/2674880d02e8b2873d48366586953f07

i should add i am quite familliar with qt and qt creator. however i have zero experience with the qt quick compiler. so any input will help

hipersayanX, 2018-02-23:
The error with the missing header is clearly some problem related to your Qt install, if you installed Qt with the online installer go to the Maintenance Tool and install the Qt/Qt 5.10.1/MSVC 2017 (or 2015 or 2013 depending on what you have installed) and MinGW.

soliton4, 2018-02-27:
i installed qt 5.10.1 just as you asked.
i configured the project with the same changes i made for 5.9.4. additional arguments; path variable
i get again the error about "private/qv4function_p.h"

can you please recheck if you have some special source distribution or your own qt sources in a special configuration?

soliton4, 2018-02-27:
i did some research and figued out that 
QT += qml-private
was missing in the project files
i added it to the akcommons.pri and to the standalone.pro

now the release build runs through

when running it i only get this output in the application output tab:

```
Starting C:\Users\paperspace\projects\webcamoid\StandAlone\webcamoid.exe...
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:27 "qrc:/Ak/share/qml/AkQmlControls": no such directory

qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_CTX_new
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
Error requesting latest version: No response
```

no window appears even after waiting a few minutes. i know webcamoid can take a long time to start up. but not that long.
task manager says it uses 0% cpu and 12.3 MB ram

hipersayanX, 2018-02-27:
I did figured out :man_facepalming:, look at this capture (note the _share_qml_AkQmlControls_AkScrollView.cpp_): 

[image]

and this error message:

```
Starting C:\Users\paperspace\projects\webcamoid\StandAlone\webcamoid.exe...
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:27 "qrc:/Ak/share/qml/AkQmlControls": no such directory
```

then going again to this capture: 

[image]

try disabling the **Enable Qt Quick Compiler** option, remove the QT += qml-private, clean the tree, and try building again, it may work now :smiley: 

Are you using the commercial version of Qt right? because it would be interesting to make it compile with the Qt Quick Compiler option enabled. AkQmlControls was actually a hacky solution to keep some features from Qt Quick Controls 1, so maybe it can be reworked to compile with Qt Quick Compiler.

soliton4, 2018-02-28:
i did as you asked and disabled the qt quick compiler.
i reset every modification i did and i deleted any files the qt quick compiler created.
it had the habbit to delete the *.qm files so i also restored those. everything is back to the state in github.

when i run it this is the output:
```
Starting C:\Users\paperspace\projects\webcamoid\StandAlone\webcamoid.exe...
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:27 "qrc:/Ak/share/qml/AkQmlControls": no such directory

qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_CTX_new
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
Error requesting latest version: No response
```
same behavior as before

i tried a clean, which delete the *.qm files again
i restored them again

made a fresh build
same result

i can compile it but i cant seem to get it to run properly
[image]

do you think these warnings have something to do with it?

hipersayanX, 2018-02-28:
I'm confused then :confused:, Webcamoid must build straightforward with vanilla Qt, no patches are required.

>i tried a clean, which delete the *.qm files again

Do you use _git clean -xfd_ to clean project tree? and _git reset --hard_ to restore to the last point?

>do you think these warnings have something to do with it?

No, the QSslSocket error is a Qt thing, because of incompatibilities between [openssl 1.0 and openssl 1.1](https://wiki.openssl.org/index.php/OpenSSL_1.1.0_Changes#Qt), and that doesn't affects Webcamoid, you only loose the ability for checking latest version and that's all, everything else must work well.
The other warnings are just stub code I'm working on, it doesn't affects build.

soliton4, 2018-02-28:
i was refering to qt clean.
i can try to clone into a new folder to make it 100% clean
i will also try to install a open source version of qt to see if that makes any difference
will post updates soon

hipersayanX, 2018-02-28:
>i can try to clone into a new folder to make it 100% clean

I don't think is necessary, just running _git clean -xfd_ and _git reset --hard_ must be more than enough.

>i was refering to qt clean.

Sometimes, running _make clean_ may not be enough since Makefiles could be tainted with old presets, that's why doing a git clean is a much better option here. Also running git clean will not delete your qm files.

hipersayanX, 2018-03-02:
Any advance? I was reading [this](https://doc.qt.io/QtQuickCompiler/) and it seems Webcamoid is almost prepared to work with Qt Quick Compiler, the only thing missing is the CONFIG += qtquickcompiler option, can you try that?

soliton4, 2018-03-02:
i can try to make a build with qtquickcompiler but first i need to get it compiling and running.
i was busy with other tasks. my plan is to make a fresh start next week
will keep you posted.

once i get it building and running i need to make some small changes. do you think there is a chance to discuss my plans and mybe get some hints? 
im getting ahead of mysel

have a good weekend

hipersayanX, 2018-03-02:
>do you think there is a chance to discuss my plans and mybe get some hints?

Of course!

>have a good weekend

Thanks, you too :smile:

soliton4, 2018-03-14:
i reinstalled everything on a new vm. i am now using the open source version of qt 5.10.1
visual studio 2017
the project is configures for vs2017 64 bit

same result as before:
```Starting C:\Users\paperspace\projects\w2\build-Webcamoid-Desktop_Qt_5_10_1_MSVC2017_64bit-Release\StandAlone\webcamoid.exe...
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:27 "qrc:/Ak/share/qml/AkQmlControls": no such directory

qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_CTX_new
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
Error requesting latest version: No response```
12.7 mb used memory. no reaction after starting it

soliton4, 2018-03-14:
you mentioned the release version is allways build under linux. have you ever tried building this project under windows?

i am a bit suspicious that you actually do the windows build under linux. can you confirm that is correct? how do you go about debugging windows specific code?

soliton4, 2018-03-14:
i managed to start a debug version of the build after doing some heavy modifications.
i had to change a lot of relative path to aboslute path

https://github.com/webcamoid/webcamoid/blob/acb0b8f9650379cc8789c8e2ecce7c5bff92e12a/StandAlone/src/mediatools.cpp#L613
the rootobjects refered to here ^ are empty

the interfaceLoaded signal is not connected to anything.
without knowing more about how the application is supposed to behave it is rather difficult to identify the errors. 

i am still offering to fix the wiki for the windows build but at this point i suspect its not only the wiki that needs fixing

hipersayanX, 2018-03-14:
Try with latest commit, now it must import the controls properly [here](https://github.com/webcamoid/webcamoid/blob/master/StandAlone/share/qml/main.qml#L27), the controls are imported as embedded resources, [this](https://github.com/webcamoid/webcamoid/blob/master/StandAlone/src/mediatools.cpp#L95) is the path where it must be located.

>you mentioned the release version is allways build under linux. have you ever tried building this project under windows?
>can you confirm that is correct? 

Yes, the release version is always build in GNU/Linux with MinGW, but also make the tests in real Windows.

>how do you go about debugging windows specific code?

In Windows with GDB and CDB, also MinGW GDB in wine.

>he rootobjects refered to here ^ are empty

Is empty because, of the import error above

```
qrc:/Webcamoid/share/qml/main.qml:27 "qrc:/Ak/share/qml/AkQmlControls": no such directory
```

The root object is ApplicationWindow in StandAlone/share/qml/main.qml.

soliton4, 2018-03-14:
the output after the latest change looks like this:
```
Starting C:\Users\paperspace\projects\w3\build-Webcamoid-Desktop_Qt_5_10_1_MSVC2017_64bit-Release\StandAlone\webcamoid.exe...
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:24 module "AkQml" is not installed
qrc:/Webcamoid/share/qml/main.qml:27 module "AkQmlControls" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "AkQml" is not installed
qrc:/Webcamoid/share/qml/main.qml:27 module "AkQmlControls" is not installed

qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_CTX_new
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
Error requesting latest version: No response
```
i did a fresh clone
only difference to the wiki is that i pass absolute path in the commandline
without the absolute path in the commandline i get 12.8 mb usage
with absolute path >16mb memory is used

i will now attempt to debug and post the results here

hipersayanX, 2018-03-14:
```
Starting C:\Users\paperspace\projects\w3\build-Webcamoid-Desktop_Qt_5_10_1_MSVC2017_64bit-Release\StandAlone\webcamoid.exe..
```
Again, don't enable shadow build.

Also in _Build & Run > Run > Command line argumments_ write 

```
-q C:\Full\path\to\webcamoid\libAvKys -r -p C:\Full\path\to\webcamoid\libAvKys\Plugins
```

soliton4, 2018-03-14:
ok but without a shadow build it is impossible to switch between release and debug
i can not do a clean because the translation files get deleted when i clean
i could do 2 folders but then i have a hard time syncing them when i do changes

hipersayanX, 2018-03-14:
>ok but without a shadow build it is impossible to switch between release and debug

Don't worry now, that can be fixed later once you get this working.

>get deleted when i clean

As I said before, use git clean -xfd instead.

soliton4, 2018-03-14:
finally:
[image]

hipersayanX, 2018-03-14:
Oh I actually did figured it out, you were using

```
-q %CD%\libAvKys 
```

instead of

```
-q C:\Full\path\to\webcamoid\libAvKys 
```

Qt Creator doesn't interpret Batch wildcards, that's why it was not loading AkQmlControls, because AkQmlControls is inside AkQml, and AkQml was not loaded because it was in wrong path.
Try setting the proper path, and it must work now :smiley:

hipersayanX, 2018-03-14:
Good!

soliton4, 2018-03-14:
about the startup parameter. i was using the absolute pathe in there for a while now
but it didnt work with the shadow build

soliton4, 2018-03-14:
thank you a lot for helping me with the compiling. i could update the wiki accoring to my findings. but i would actually prefer making this project compatible with a shadow build first. 

i noticed that the debug build is configured to create the translations. whats up with that?

do you think there need to be a lot of changes to get a shadow build running?

soliton4, 2018-03-14:
wrong password:
[image]
when i try to create a camera i get "wrong password" is this a problem of my build?

hipersayanX, 2018-03-14:
>i would actually prefer making this project compatible with a shadow build first.

Yes, please.
Also, it would be nice if you can make it work with Qt Quick Compiler.

>i noticed that the debug build is configured to create the translations. whats up with that?

qm files must exists before build and specified in qrc file otherwise qmake will not be embed them.

>when i try to create a camera i get "wrong password" is this a problem of my build?

Don't worry, is because the code to install the new virtual camera is not there yet.
You can install the DLL filter in webcamoid/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCamera with [GraphStudioNext](https://github.com/cplussharp/graph-studio-next).

soliton4, 2018-03-14:
> Yes, please.
cool
> Also, it would be nice if you can make it work with Qt Quick Compiler.
how about some additional instruktions for owners of a licensed qt version?

> qm files must exists before build and specified in qrc file otherwise qmake will not be embed them.
i believe this is true for debug as well as for release versions. however this:
https://github.com/webcamoid/webcamoid/blob/abd04f4597a4161e4f277a97961ca106ba72d42a/commons.pri#L130
only affects debug builds
since it is the only reference to TRANSLATIONS_PRI that i found, i assume this is the code to create the translations? i am not an expert in qml and translations. (never used them)

> Don't worry, is because the code to install the new virtual camera is not there yet.
does this mean this code is about to change? i need the virtual camera feature. can i use the old version of this code? or will you finish your modifications within the next few days?

GraphStudioNext (?!) not another complicated tool to get into - i was so happy to find your project so i can use the virtual camera feature out of the box without thinking to much about microsoft apis ;)

hipersayanX, 2018-03-14:
>how about some additional instruktions for owners of a licensed qt version?

If QQC still ask for private headers you can check if CONFIG += qtquickcompiler is used and then add the QT += qml-private you used before, that way open source and commercial editions of Qt will be 1:1 and no special instructions may be required, maybe.

>i believe this is true for debug as well as for release versions. however this:...

What it say there is that translations will be updated in debug build only, and you can disable it if you don't want, that's because GNU/Linux distro packagers will normally just take the source compile and make  the package. Not compiling the translations in release will speedup the build for them.

The [TRANSLATIONS_PRI](https://github.com/webcamoid/webcamoid/blob/master/StandAlone/StandAlone.pro#L19) variable is controlled by [translations.pri](https://github.com/webcamoid/webcamoid/blob/master/translations.pri) file, this file scans for all string sources in C++ and Qml files and deposit the sources in _StandAlone/share/ts_. The translations will only be triggered when compiling the StandAlone.

>does this mean this code is about to change? 

Yes I'm working in this right now.

>i need the virtual camera feature. can i use the old version of this code?

Maybe?

>or will you finish your modifications within the next few days?

In few days pretty sure not, but the new virtual camera will be much better than the older one, I been commenting some things about that [here](https://github.com/webcamoid/webcamoid/issues/90#issuecomment-372725869) and [there](https://github.com/webcamoid/webcamoid/issues/112#issuecomment-368580479). I want a 1:1 implementation with Mac virtual camera and the future GNU/Linux virtual camera.

>GraphStudioNext (?!) not another complicated tool to get into

This was actually a quite cool tool to debug the virtual camera, obviously Webcamoid will not require it once finished.

soliton4, 2018-03-15:
i updated the build instructions to work with a shadow build
i also made a pr to make the shadow build work with a debug configuration.
the release build should work with the new instructions

hipersayanX, 2018-03-15:
PR merged and added some extra tips there.

soliton4, 2018-03-15:
thanks for your help. im gonna open a new issue for the non working webcam

closed 2018-03-15 by soliton4