webcamoid/webcamoid #112: New feature, idea

ROBERT-MCDOWELL, 2018-02-21:
# New feature, idea

Hi,
I'm looking for a video sink for Mac OS and Linux where the full desktop raw frames are going to,
with possibility to choose a monitor if multi. The most simple and light weight, the video sink can be recognized from flash,HTML5,skype etc...
Do you think it's possible to split a part of webcamoid project and use it for this purpose?
thanks

hipersayanX, 2018-02-21:
Do you mean a virtual webcam? this is already in progress for Mac OS, and for GNU/Linux it's already implemented with v4l2loopback, but is not up to my standards, I will work in my own virtual camera module once I finnish with DirectShow plugin. it will be ready for 8.5.0 release.

ROBERT-MCDOWELL, 2018-02-23:
> Do you mean a virtual webcam
a kind of. a good example for windows here (http://www.umediaserver.net/components/index.html)
http://www.umediaserver.net/components/index.html
for example when you select it from any web software in flash,html5 or skype
it automatically show in raw mode the entire desktop.
I'm following V4l2loopback but it's an empty sink only and need like ffmpeg to create a stream
I actually not found such kind of thing on Mac OS for now unless to download an entire software like Webcamoid or CamTwist.

hipersayanX, 2018-02-23:
>Do you mean a virtual webcam
a kind of. a good example for windows here (Screen Capture DirectShow source filter)
http://www.umediaserver.net/components/index.html
for example when you select it from any web software in flash,html5 or skype
it automatically shows in raw mode the entire desktop.

Yes, that is what the virtual camera actually does.

>I'm following V4l2loopback since years but it's an empty sink only and need like ffmpeg to create a content

v4l2loopback is just the kernel module, it's only purpose is to communicate with the programs to send and receive the frames, that's why you need some program that configure the kernel module (output format resolution, description), actually only a very few programs make use of v4l2loopback.

>I actually not found such kind of thing on Mac OS for now unless to download an entire software like Webcamoid or CamTwist.

As explained before, you need a program that communicate with the driver/kernel module/plugin, the program can be simple or complex, but you need a program.

>I just would like to avoid to use any google webrtc solution....

Virtual camera has nothing to do with webrtc, that something other programs implements or not.

ROBERT-MCDOWELL, 2018-02-23:
Thanks for your answers.
> Virtual camera has nothing to do with webrtc, that something other programs implements or not.
I know, actually there is a google extension needed to use webrtc with desktop streaming.
btw, I don't have enough C skills to analyze how UscreenCapture is grabbing the screen into its virtual device but at least it's a pretty small program (less than 200k).

ROBERT-MCDOWELL, 2018-02-23:
> Believe me, I've made a lot of research i
why do you think I don't believe you? :\
I certainly know that screenshot from desktop is tied to the OS, I'm not a newbie on this matter,
but certainly ignorant of how to make something heterogeneous for the end user.

hipersayanX, 2018-02-23:
>why do you think I don't believe you? :)

It's only an expression, sorry :P

ROBERT-MCDOWELL, 2018-02-26:
I'm kidding ;).
ok, so is webcamoid includes this virtual video device in its source already?

hipersayanX, 2018-02-26:
Yes, Webcamoid is the GUI to the virtual camera, that's why it make no sense to create another program to do the same. Just in theory you can create a GUI independent to Webcamoid and interfacing with the driver (in v4l2loop it's already like that) using any GUI you want, since the only dependency is the system itself, but I can't guarantee API compatibility with my own drivers.
Oh, i forgot to mention before, but yes, Webcamoid can actually capture from multiple monitors (or may be) and send the frames to the virtual camera. That's why I consider this feature already done.

ROBERT-MCDOWELL, 2018-02-26:
ok gotcha, but on Mac OS and Windows what kind of sink are you using?
do you think it's possible to make webcamoid react like UscreenCapture,
like if I select the webcamoid sink as video source it will automatically launch webcamoid in background?

hipersayanX, 2018-02-26:
>but on Mac OS and Windows what kind of sink are you using?

GNU/Linux: v4l2loopback, but I will write my own after finishing with Windows driver.
Mac: CoreMediaIO in development branch, it will replace Syphon that is used in 8.1.0 and did not work as expected.
Windows: DirectShow, I rewriting it right now to fix some problems with the old driver. I will be reporting the advance later in issue #90.

>do you think it's possible to make webcamoid react like UscreenCapture, like if I select the webcamoid sink as video source it will automatically launch webcamoid in background?

TBH I was about to say no, but I think it could be possible using a daemon that communicate with the driver and launch webcamoid.

>are you using GTK for the GUI?

Webcamoid uses Qt, it was the only valid option for portability, and because Qt interfaces looks great! :smile:

ROBERT-MCDOWELL, 2018-02-26:
good to know, useful. I would like to contribute but my C skills are not enough strong to not create a mess.
but for sure I can run your alpha beta dev in intensive environment very quickly though

hipersayanX, 2018-02-27:
>direct show still available on win 10?

Yes, but making virtual webcam work with Windows Store apps is quite trick, I will give more details later in issue #90.

>I don't know how UscreenCapture does it, but it works very well and transparently without to burden the system. you can take a look at the source or installer, maybe it will inspire you.

Not necessary, I already know how can be done, and UscreenCapture is closed source anyways.

ROBERT-MCDOWELL, 2018-02-27:
ok thanks for all the info, if you need anything just tell me, hardware etc... I will do my best to help you...