webcamoid/webcamoid #113: RTP source

petergerten, 2018-02-25:
Hi,

I could not find any details on what kind of sources are possible. Obviously I can use a camera or a file source and searching this repo, it seems RTSP sources work as well. Is there any complete source list? I am particular interested in a direct RTP source.

Peter

hipersayanX, 2018-02-26:
Yes, RTSP are valid sources it may recognize any URL you throw at it, although I've not tested it thoroughly. In theory it may support any format supported by FFmpeg, such as local and remote files, URL streams, GIF, and maybe other formats,

Y2Kot, 2020-01-28:
Is there any documentation about how to connect rtsp stream to webcamoid? I'm using windows and can't create source with rtsp from UI

hipersayanX, 2020-01-28:
https://github.com/webcamoid/webcamoid/issues/213#issuecomment-571705169