webcamoid/webcamoid #114: Question about the dshow virtual camera sendFrame()

andychan1998, 2018-02-27:
I am reading the webcamoid virtual camera code in order to understand how it works with directshow.  Inside the pin.cpp's sendFrame() function, it simply copies over random bytes:

    for (LONG i = 0; i < size; i++)
        buffer[i] = rand() % 255;

I couldn't quite figure out how the directshow filter obtains the video stream input and sends it to the application.  I would really appreciate some help in understanding the code.  Thank you very much.

hipersayanX, 2018-02-28:
Just in case, the code you see there is the code for the new virtual webcam I'm writing right now, the code for transferring the frames is not there yet. 
If you want to know how transfer the frames you can look at current release code, [right here](https://github.com/webcamoid/webcamoid/blob/28a7b62bc589c8f8452af70bd941156249ca34da/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCameraFilter/VirtualCameraSource/src/virtualcamerasourcestream.cpp#L213), readBitmap can either read from a resource (if Webcamoid is not sending frames) or from the [ipcBridge](https://github.com/webcamoid/webcamoid/blob/28a7b62bc589c8f8452af70bd941156249ca34da/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCameraFilter/VirtualCameraSource/src/virtualcamerasourcestream.cpp#L909), and what IpcBridge::read does is reading the frames from a [file map](https://github.com/webcamoid/webcamoid/blob/28a7b62bc589c8f8452af70bd941156249ca34da/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCameraFilter/ipc/src/ipcbridge.cpp#L143).
But I don't like that approach, I want to do something similar to what I did with Mac virtual camera, that is using the GPU to transfer the frames using [Surface Sharing](https://msdn.microsoft.com/en-us/library/windows/desktop/ee913554(v=vs.85).aspx), and for message passing I will probably use a [COM Server](https://msdn.microsoft.com/en-us/library/windows/desktop/ms683835(v=vs.85).aspx), so the virtual webcam can work with both 32 and 64 bits, this is the nearest equivalent to IOSurface and XPC in Mac.

andychan1998, 2018-02-28:
Thanks for the quick response.  I'll take a look.

hipersayanX, 2018-07-05:
Question answered so closing.

closed 2018-07-05 by hipersayanX