webcamoid/webcamoid #116: add webcam broken

soliton4, 2018-03-15:
### Summary

add webcam results in "wrong password" error

### Current Behavior

building the current sources the add webcam function seems broken. you mentioned in #111 that this is due to code in change. 

### Expected Behavior

the webcam should have been created and work

### Steps to Reproduce

build from the current master branch and try to add a camera

### Suggestions and tips

could you do the breaking changes in a branch so that the master branch is still usable?
alternatively could you point to a commit where it was still working? i mean before you introduced your breaking change

### Your Environment

* Webcamoid information: latest version from master branch
* Operating System information: windows server 2012

hipersayanX, 2018-03-15:
The latest commit for the old virtual webcam was 805dedf503e6df84d5630d989bb5e2293c5e9714, you will find all files in _webcamoid/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCameraFilter_.

soliton4, 2018-03-15:
thanks. i should be able to branch off from there and cherry-pick 
https://github.com/webcamoid/webcamoid/commit/760779425a399c56344a2994a10346000a949643

to make it compile on my windows system. any objections?

is there any information about the nature of the change in the virtual camera? i tried the downloadable version and it worked great in google hangouts

hipersayanX, 2018-03-15:
>to make it compile on my windows system. any objections?

No problem, Webcamoid is Free Software so you can do whatever you want with it while complying with GPLv3+.

>is there any information about the nature of the change in the virtual camera?

A lot. Most of the reasons listed here #90:

- The problem with coexisting 32 bits and 64 bits apps in Windows 64 bits,  ie. the 64 bits camera doesn't works in 32 bits apps.
- The video formats and resolutions are fixed.
- Only one available camera.
- It's not designed to receive frames from multiple instances of Webcamoid.
- Build constantly broking with every new MSVC release.
- Unclear BaseClasses license, now it's licensed under MIT, but in some parts of Microsoft website it still say sample code is licensed under Ms-LPL (which is not Free at all), while in the old sample code downloads does not show any license file, plus the "M$ all rights reserved" in all code headers.
- Horribly structured spaghetti code, mixing several classes in one file.

soliton4, 2018-03-15:
ok wasnt aware of all those problems.
do you have any idea when the new solution will work?
no pressure just asking

hipersayanX, 2018-03-15:
Mac camera took me 2 months to complete, I've started 1 month ago with this and it's in the half part (the other half is writing the assistant and IPC), so maybe 1~1.5 month.

soliton4, 2018-03-16:
i rolled back to the mentioned commit (see screenshot) but i still get the same error (see screenshot)

[image]

soliton4, 2018-03-16:
it looks like some of the compile results were not marked as dirty when i did the rollback. here is the result after deleting the shadow build folder:

[image]

i noticed the same code is in place for the release 8.1.0 tag

did this ever compile on windows 64bit?

hipersayanX, 2018-03-16:
Yep, and that's exactly what I said before:

>Build constantly broking with every new MSVC release.

The code you are pointing out there is Microsoft code, I tried to patch several times, when released 8.1.0 it compiled well, then broke again sometime later, and I did not touch anything.

If you want to continue from there (compiling with MSVC), you can try patching yourself, I really don't want to see that code anymore. The other two options are, if you don't care about making the virtual camera work in 64 bits programs, you can compile with Qt's builtin MinGW 32, the other option is using [MSYS2](https://www.msys2.org/) which has a 64 bits version for all required libraries. The virtual camera may compile flawless with MinGW, at least the last time I tried it worked.

hipersayanX, 2018-03-17:
>but can i still contact you once i get it compiling if i run into problems?

With Webcamoid code of course, but I will not take responsibility on Microsoft's horrendous code.

>the idea is to make webcamoid build as dll and use that in my application.

Don't over complicate, if you want to send the frames directly to the webcam, a lot much simpler solution, since the release version camera is working, is compiling the [ipcbridge part](https://github.com/webcamoid/webcamoid/tree/805dedf503e6df84d5630d989bb5e2293c5e9714/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCameraFilter/ipc) and linking it to your project. A basic example for sending frames would be:

```
IpcBridge bridge(IPC_FILE_NAME);

if (bridge.open(Write)) {

    for (;;) {
        BYTE data[3 * 640 * 480];

        // Fill data.

        bridge.write(MEDIASUBTYPE_RGB24, width, height, data);
    }

    bridge.close()
}
```

>as far as i know that would be ok with the gpl. can you confirm?

Programs linking to GPL code and released publicly must be GPL, except when your code is for private usage (ie. making some project you don't plan releasing to the world).

soliton4, 2018-03-22:
@hipersayanX that worked pretty well. i am a total noob when it comes to windows IPC (well maybe not total but you dont use memory mapped files anymore do you?)
but i was able to get it working out of the box

is there also a way to feed audio?

hipersayanX, 2018-03-22:
>you dont use memory mapped files anymore do you?

In new implementation, no. But I'm experimenting with pipes for message passing. and I did not any research yet but will probably use Surface Sharing for transferring frames.

>is there also a way to feed audio?

No yet, but it's planned after adding support for audio effects, maybe for 10.x or 11.x series, anyways 9.x series will be a long long ride.

hipersayanX, 2018-03-22:
>do you have any tips for how to write an audio input driver that works similar to your video input driver?

Check Graph Studio Next for audio sources, which category and interfaces it implements and implement those category and interfaces into your filter.

>is there a way to call webcamoid to install the webcam from the command line without user interaction?

Not yet, but may not be hard to add that option.

>is there a way to check if the camera is installed?

[Check the ClassID](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/dshow/src/cameraoutdshow.cpp#L98) of the enumerated cameras. For the old camera it is:

```
DEFINE_GUID(CLSID_VirtualCameraSource, 0x41764b79, 0x7320, 0x5669, 0x72, 0x74, 0x75, 0x61, 0x6c, 0x43, 0x61, 0x6d);
```

hipersayanX, 2018-07-05:
[wiki](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Windows) updated, and the new virtual webcam code is more or less stable now. Current changes are:

- Video formats, resolutions and frame rates can be defined dynamically. [1]
- Webcam description can be written in any language.
- Options like mirror, scaling smoothness, aspect ratio and RGB swapping can be controlled from Webcamoid. [2]
- Video frames can be routed to many clients and from one or many instances of Webcamoid.
- Connection can be recovered in case some peer die (Webcamoid or a client).
- Frames can be shared between any 32 and 64 bits peer (Webcamoid or a client).

[1] Will define some common formats in the code before release. Read [2].
[2] With time running, I'm pretty sure will be not able to map these options to the GUI, I will add them for Webcamoid 9.0.0.

ROBERT-MCDOWELL, 2018-07-05:
hipersayanX:
I cannot see any donation button on your website, maybe it would be worth to add one?

hipersayanX, 2018-07-05:
@ROBERT-MCDOWELL no yet, but soon :wink: , also thank!

hipersayanX, 2018-12-27:
This was already solved in master branch.

closed 2018-12-27 by hipersayanX