webcamoid/webcamoid #117: Window opens hidden on another monitor. ( Windows 64) multiple monitor

KHappe, 2018-04-03:
This is similar to issue #101 ( The application window does not appear),
But on a Windows multiple monitor setup.
I have a windows 7 laptop with 3 monitors. When I launch the application, I get the open animation but it  never appears.  It sort of seems like it going to a monitor that doesnt exist.

When I undock my laptop.  The app does appear.  I was hoping that it would fix the issue but it doesnt it.

hipersayanX, 2018-04-03:
>This is similar to issue #101 ( The application window does not appear)

Not at all, the other issue is about missing dependencies when packaged in GNU/Linux distributions.
In your case I think it could be a Qt bug. I'm not sure if I will able to fix this, I only have one old VGA monitor to test. Some more information? are you using VGA or HDMI output? are you using an splitter?

adrianhoehne, 2021-05-29:
Setup: 
Windows 10 Pro 20H2 Multimonitor with GTX970 and Dockingstation. 3 Monitors on GTX and 2 on Docking Station
App Version 8.8.0

Hi, in my case it every time wants to open on screen 1 regardless that the main screen setup on window is 2. Also if I resize the window or maximize it.