webcamoid/webcamoid #118: After downloading webcamoid

uselessone, 2018-04-26:
Hey guys i downloaded webcamoid then i see the issue.My webcam working normal before download that app and quality was good i could see  myself but after downloading my i cant see anything and and poor video  quality tried configure but it didnt help please feed back.

hipersayanX, 2018-04-26:
You can choose resolution and format in the right side bar, Webcamoid does not modify nothing at global level, so it must not affect other programs. Also, which operating system?

hipersayanX, 2018-04-26:
Oh wait, I remember, brightness, contrast and all those features are affected globally (resolution and format not). Could be that you are using Windows?

hipersayanX, 2018-12-27:
This seems to be the same issue as #148, will continue there since it has more information.

closed 2018-12-27 by hipersayanX