webcamoid/webcamoid #121: linux : Virtual Desktop webcam  : error  v4l2

Sylvain-07, 2018-05-21:
Virtual Webcam Desktop don't work's 

error in my terminal :
[code][sylvain-sa@SYLVAIN-SA Desktop]$ webcamoid 
libv4l2: error getting pixformat: Argument invalide VideoCapture: Can't set format: "YUYV" [/code]

[sylvain-sa@SYLVAIN-SA Desktop]$ webcamoid -v
Webcamoid 8.5.0
[sylvain-sa@SYLVAIN-SA Desktop]$ ffmpeg -version
ffmpeg version N-91132-gcadf7a7f39 Copyright (c) 2000-2018 the FFmpeg developers
built with gcc 8.1.0 (GCC)
configuration: --prefix=/usr --extra-cflags=-I/opt/cuda/include --extra-ldflags=-L/opt/cuda/lib64 --extra-libs=-lpthread --disable-rpath --enable-gpl --enable-version3 --enable-nonfree --enable-shared --disable-static --enable-gray --enable-avresample --enable-alsa --enable-avisynth --enable-bzlib --enable-chromaprint --enable-frei0r --enable-gcrypt --enable-gmp --enable-gnutls --enable-iconv --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcelt --enable-libcdio --enable-libcodec2 --enable-libdc1394 --enable-libfdk-aac --enable-libflite --enable-fontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libiec61883 --enable-libilbc --enable-libjack --enable-libkvazaar --enable-libmodplug --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --disable-libopencv --enable-libopenh264 --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-librtmp --enable-libshine --enable-libsmbclient --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtesseract --enable-libtheora --disable-libtls --enable-libtwolame --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvo-amrwbenc --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxavs --enable-libxcb --enable-libxcb-shm --enable-libxcb-xfixes --enable-libxcb-shape --enable-libxvid --enable-libxml2 --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-lzma --enable-decklink --enable-libndi_newtek --enable-libmysofa --enable-openal --enable-opencl --enable-opengl --disable-openssl --enable-sndio --enable-sdl2 --enable-vapoursynth --enable-xlib --enable-zlib --enable-cuda-sdk --enable-cuvid --enable-ffnvcodec --enable-libdrm --enable-libmfx --enable-libnpp --enable-nvdec --enable-nvenc --enable-omx --enable-omx-rpi --enable-rkmpp --enable-v4l2-m2m --enable-vaapi --enable-vdpau
libavutil      56. 18.102 / 56. 18.102
libavcodec     58. 19.102 / 58. 19.102
libavformat    58. 15.100 / 58. 15.100
libavdevice    58.  4.100 / 58.  4.100
libavfilter     7. 24.100 /  7. 24.100
libavresample   4.  0.  0 /  4.  0.  0
libswscale      5.  2.100 /  5.  2.100
libswresample   3.  2.100 /  3.  2.100
libpostproc    55.  2.100 / 55.  2.100

[sylvain-sa@SYLVAIN-SA Desktop]$ v4l2-ctl --list-device
camoid (platform:v4l2loopback-000):
	/dev/video2

Webcam C170: Webcam C170 (usb-0000:00:1a.0-1.4):
	/dev/video0
	/dev/video1

Sylvain-07, 2018-05-26:
It's not work since i have upgrade my kernel to 4.16.9
before it work correctly.
i have reinstal ffmpeg, v4l2, webcamoid :  not solve.
Since this upgrade my web cam has 2 dev : 
/dev/video0 and /dev/video1 : it's the same webcam. : before /dev/video0

hipersayanX, 2018-05-26:
It seems to be a bug in v4l2loopback it self, [here](https://github.com/umlaeute/v4l2loopback/issues/172) is the related issue, you may report the bug there, there is a lot of people with the same exact issue.
For me, I'm about to finish writing the virtual camera for Windows, and start writing my own virtual camera for Linux and remove v4l2loopback as dependency.

Sylvain-07, 2018-05-28:
Today : 
i have upgraded to kenel 4.16.11
recompile v4l2loopback-dkms-git 
to patch soluce : https://github.com/umlaeute/v4l2loopback/blob/master/v4l2loopback.c#L2096
I have recompile webcamoid

manually  Virtual Desktop webcam work correctly witch 
ffmpeg -f x11grab -r 15 -s 1920x1080  -i :0.0+0,0 -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video2

but it's alway's out witch webcamoid

hipersayanX, 2018-08-06:
Some updates on virtual camera for Linux. From my previous comment:

> For me, I'm about to finish writing the virtual camera for Windows, and start writing my own virtual camera for Linux and remove v4l2loopback as dependency.

I'm actually planning to keep support for v4l2loopback (for high performance), but at same I'm experimenting with [fuse](https://github.com/libfuse/libfuse) to create a user space virtual camera, may be much easier to maintain while giving support for other POSIX compliant operating systems for free.

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

closed 2019-01-22 by hipersayanX