webcamoid/webcamoid #126: More Install/run issues on MacOS High Sierra

piotrm0, 2018-06-23:
Both the master or the devel branches on MacOS High Sierra have an issue when attempting to follow the [MacOS build instructions](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Mac). Building goes through without major issue but running results in:

```
/Users/piotrm/Dropbox/repos/github/deps/webcamoid/StandAlone/debug/Qt5.11.1/clang/bin/webcamoid.app/Contents/MacOS/webcamoid...
QML debugging is enabled. Only use this in a safe environment.
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:24 module "AkQml" is not installed
qrc:/Webcamoid/share/qml/main.qml:27 module "AkQmlControls" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "AkQml" is not installed
qrc:/Webcamoid/share/qml/main.qml:27 module "AkQmlControls" is not installe
```

This is similar to an earlier Linux issue #101 in that the contents of `libAvKys/AkQml` is unavailable. In that issue it is noted that these things need to be installed to `/usr/local/Cellar/qt/.../qml` but the qmake generated Makefile in libAvKys/AkQml attempts to install things into `/usr/lib/qt` . In High Sierra this directory is no longer writable so even that fails.  

Manually moving the built `AkQml/libAkQml.dylib` to the correct cellar location solves the run error message but has plenty of other issues afterwards most likely due to other things (plugins, others?) not being in the correct place. I don't know how to adjust the project settings or the build instructions to make sure all necessary requirements are placed correctly. Any help with that would be appreciated.

piotrm0, 2018-06-23:
I have been making some progress by making manual adjustments to paths given to `qmake` as described in the [raw paths](https://github.com/webcamoid/webcamoid/wiki/Raw-build-and-install#install-paths) document. The AkQml install bath can be adjusted by providing the correct `INSTALLQMLDIR` as noted in #101 when constructing 
`libAvKys/AkQml/Makefile`. However, the other components like `libAvKys` and `libAvKys/Plugins` have different path issues and need other paths adjusted. Given how manual this process is turning out to be I must not be doing things the right way.

hipersayanX, 2018-06-23:
I need to update the documentation for GNU/Linux, Mac and Windows, but I had no time because I was focusing all the development in the virtual camera.

Now about the issue, because development paths and install paths differs, you can specify where Webcamoid must find the Qml file, plugins and virtual camera, you can do so in Qt Creator in _Projects > Build & Run > Command line argumments_. And in your case for example you must add:

```
-q /Users/piotrm/Dropbox/repos/github/deps/webcamoid/libAvKys -r -p /Users/piotrm/Dropbox/repos/github/deps/webcamoid/libAvKys/Plugins --vcam /Users/piotrm/Dropbox/repos/github/deps/webcamoid/libAvKys/VirtualCamera/src/cmio/VirtualCamera
```

you can find all the command line options in the [manpage](https://github.com/webcamoid/webcamoid/blob/master/StandAlone/share/man/man1/webcamoid.1) and [clioptions](https://github.com/webcamoid/webcamoid/blob/master/StandAlone/src/clioptions.cpp#L27).

ROBERT-MCDOWELL, 2018-06-23:
hipersayanX
maybe we can help you to update the doc?

hipersayanX, 2018-06-23:
@ROBERT-MCDOWELL yes, please :pray:

ROBERT-MCDOWELL, 2018-06-23:
if you have already some texts to cleanup or else so
just put them in a raw mode or dunno if you have another work method

hipersayanX, 2018-06-23:
I have not any kind of raw documentation, but when people ask me things [like in this issue](https://github.com/webcamoid/webcamoid/issues/126#issuecomment-399693669), I use that information to edit the wiki. Probably a good start point would be the [Documentation tag](https://github.com/webcamoid/webcamoid/labels/Type%3A%20Documentation).

ROBERT-MCDOWELL, 2018-06-24:
ok so other followers please be conscious that develop a project like webcamoid is a big thing so it would be nice if everyone can give a hand on it. I'm going to do it.

ROBERT-MCDOWELL, 2018-06-24:
Also please hipersayanX provide all the wiki inks where we can modify it.

hipersayanX, 2018-06-24:
Right now the [wiki](https://github.com/webcamoid/webcamoid/wiki) can be edited by everyone, so if someone found a missing information they can add it, if there doubts just ask me.

hipersayanX, 2018-07-05:
[wiki](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Mac) updated.

closed 2018-07-05 by hipersayanX