webcamoid/webcamoid #129: feedback from version 8.1.0

ROBERT-MCDOWELL, 2018-07-06:
### Summary
everything works fine but Edge and IE on windows 10 64bits pro

### Current Behavior
the virtual camera is never shown

### Expected Behavior
to work on Edge and IE on windows 10 64bits

### Steps to Reproduce

1.  Open Edge or IE
2.  Choose a Video source from Flash or HTML5

### Suggestions and tips
# this is my opinion, it's up to you to consider it as valid or not
- optimize the left menu text disposition (everything is centered and long texts have the right side hidden)
- Cancel button from the file manager when record is clicked

# Want a new feature?
- When a virtual camera is created, how to link it with Desktop screen as source?
- when the user choose webcamoid as source in a web application so it launches hddien in background (without the GUI with last settings loaded?)
UScreenCapture from Unreal does it, it does not need to launch anything first, it's running when the web application select it as video source.

hipersayanX, 2018-07-07:
Microsoft Edge is UWP app, UWP apps only implements a [very limited set of DirectShow](https://msdn.microsoft.com/en-us/library/windows/apps/mt592898.aspx#directshow_10) and non of them are related to the webcam, and the virtual webcam is implemented as DirectShowFilter, so there is no way to make it work there. UWP apps use [MMF to capture from webcam](https://docs.microsoft.com/en-us/windows/desktop/medfound/audio-video-capture-in-media-foundation), the problem is that MMF doesn't implements any API for writing a virtual camera device, so it leaves only one possible solution that is implementing the virtual camera as a driver, just as explained [here](http://alax.info/blog/1722), and since Windows 8.0 or so drivers must be signed to work, and signing means :dollar: plus other requisites I'm not sure about, maybe signing a NDA for not disclosing the source code?

The camera app in Window 10 and Skype from Windows Store also have this problem.
About IE, not sure, did not tested. Will add a compatibility table to the wiki.

>optimize the left menu text disposition 

Definitely yes, and actually the whole GUI need a redesign.

>Cancel button from the file manager window does not work when record is clicked

This is not ok, will check, any way bugs like this must be reported separately, otherwise I will forget.

>When a virtual camera is created, how to link it with Desktop screen as source?

Just enable the vcam and play the desktop.

>when the user choose webcamoid as source in a web application so it launches hddien in background (without the GUI with last settings loaded?)

Yes it can be done, what you say is basically [this feature](https://github.com/webcamoid/webcamoid/issues/112), but it will need some redesign in Webcamoid, Don't worry I'm actually taking in consideration all this reports for the redesign.

ROBERT-MCDOWELL, 2018-07-07:
> Windows 8.0 or so drivers must be signed to work, and signing means 💵 
just create a page here with a button donation so users would be happy to contribute,
especially if it's to make it work on Edge and IE on windows 10, not a lot of virtual cam are working
on these browsers...

> Yes it can be done, what you say is basically this feature, 
sorry I forgot I already mentioned it, yes this feature is not really for me, but for users who do not really understand anything about virtual camera settings etc...

> Definitely yes, and actually the whole GUI need a redesign.
if any is an application designer please help him thanks

I open a new issue for the cancel button

hipersayanX, 2018-07-07:
>just create a page here with a button donation so users would be happy to contribute,

Is not that easy, I need to go into all the bureaucracy of creating a bank account, and I need to be sure that donations will be enough to at least pay the account and required taxes monthly, that's why I'm delaying it. Also, I must think if plain donation will be ok, or use some reward system like Patreon or Liberapay, to encourage people to donate.

ROBERT-MCDOWELL, 2018-07-09:
a link talking about how M$ crippled UWP
https://answers.microsoft.com/en-us/skype/forum/skype_win10-skype_callms-skype_audioms/since-all-non-actual-webcam-video-devices-are-not/546b4453-f64d-432b-abbf-3a4116d49a73

hipersayanX, 2018-07-10:
Quite fun the workaround :laughing: , other option would be running Windows in VirtualBox and using the Linux virtual camera.
Also did you tried the camera app that come with Windows 10? that thing does not even work, at least not on my case and many people reporting the same issue.

ROBERT-MCDOWELL, 2018-07-10:
Never, I even didn't know they have an app called camera app! :)
anyhow, I don't trust their softwares, especially the one that use my webcam...

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

closed 2019-01-22 by hipersayanX