webcamoid/webcamoid #133: Idea to lower cpu when stream desktop

ROBERT-MCDOWELL, 2018-07-10:
VNC uses mirror driver to optimize the screen refresh performance.
uvnc did a small app that record the desktop using this mirror screen driver dividing the cpu by 3 to 5.
maybe is there a way to bridge webcamoid with these mirror drivers and avoid to refresh the whole desktop every frame?

hipersayanX, 2018-07-10:
Not sure how they does, must check the sources, but most probably is something closely related to the graphics server. I prefer to see how they does and copy the functionality into Webcamoid, instead of adding VNC support just for desktop capture. I don't like to add third party dependencies for no reason.

ROBERT-MCDOWELL, 2018-07-10:
here is the link of unvc screenrecorder and their sources
http://www.uvnc.com/screenrecorder/

ROBERT-MCDOWELL, 2018-07-10:
I tried on windows XP,vista,7,10 and works well. CPU is around 13/40 % which is pretty amazing on 
a 1920x1200 screen.