webcamoid/webcamoid #135: virtual video sink on windows

ROBERT-MCDOWELL, 2018-07-10:
hipersayanX:
is it possible to separate your code that create a virtual video device into a standalone application
so this media sink can be used like a output pipe?
thanks

hipersayanX, 2018-07-10:
Possible yes, but I don't feel like maintaining two apps. But for anyone interested, they can start looking at [VCamIPC](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/dshow/VCamIPC) that is the code that controls the virtual camera.

ROBERT-MCDOWELL, 2018-07-10:
Well, I think it's time for me to learn more about C.... ;)
if the browsers restrict more and more virtual devices recognition on browsers,
as I know on computer world nothing is impossible, so anyone should investigate a workaround....

hipersayanX, 2018-07-10:
>Well, I think it's time for me to learn more about C.... ;)

https://upload.wikimedia.org/wikipedia/commons/e/e9/CPlusPlusProgramming.pdf
https://doc.qt.io/qt-5/gettingstarted.html

:smile: 

>if the browsers restrict more and more virtual devices recognition on browsers,

Well at least Firefox is more or less collaborative, Flash seems to be doomed, and Apple and Microsoft go on their own as always.

ROBERT-MCDOWELL, 2018-07-10:
didn't try webcamoid on linux yet, but your driver can be shown on all browsers but Edge and IE on windows 10.
maybe a crazy idea, there are usb dongle emulators here
http://www.dongleservice.com/emulate-hasp.phtml
dunno if it can emulate a video devices though....
despite I a C novice, I'm a programmer since 30 years though :D

ROBERT-MCDOWELL, 2018-07-11:
another interesting link about USB emulator
http://usbip.sourceforge.net/

ROBERT-MCDOWELL, 2018-07-11:
also here:
https://github.com/Microsoft/UDE/tree/master/UDEMbimClientSample

hipersayanX, 2018-07-11:
In the case of Windows, creating an AVStream driver seems to be the right way:

https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/avstream-overview
https://docs.microsoft.com/en-us/windows-hardware/drivers/samples/avstream-driver-samples
https://github.com/Microsoft/Windows-driver-samples/tree/master/avstream

ROBERT-MCDOWELL, 2018-08-09:
Also, OBS project has a lot of good code to be inspired from
https://github.com/obsproject/obs-studio

ROBERT-MCDOWELL, 2018-10-17:
there is also an interesting project called droidcam whic has very good virtual video devices seen by all browsers and versions, maybe it's a good source of inspiration
https://github.com/aramg/droidcam

hipersayanX, 2018-10-17:
The Linux virtual camera turned to be quite challenging, hopefully maybe it will be ready in less than a month, then I will integrate in Webcamoid, debug all the thing and release :sweat_smile:
I want to make a new release before the end of this year, so any other plan for this release will totally ignored :x:

ROBERT-MCDOWELL, 2018-10-17:
ok no worries, is v4l2loopback too bad to be used on Linux?

ROBERT-MCDOWELL, 2018-10-18:
Ha indeed it's obviously relevant :)

closed 2018-12-27 by hipersayanX