webcamoid/webcamoid #137: Question about real time montage

ROBERT-MCDOWELL, 2018-08-03:
# Reporting a problem?
NO

# Want a new feature?
Not really, I just wonder if it's possible to make an easy montage
like this video, mixed webcam, background and screenshare in realtime
https://www.youtube.com/watch?v=7ZPDQf_dO94
and if it's possible to set this config on a file.
thanks

hipersayanX, 2018-08-04:
Do you mean picture in picture, adding banners and logos? this is actually a planned feature, my idea is to support different profiles, for example for screencasting, reviews, gameplays, and so on. This is basically the one of the features that will define 10.x series.

ROBERT-MCDOWELL, 2018-08-04:
mostly side by side webcam and screenshare with free to position and size
a movie or image in background