webcamoid/webcamoid #138: Random effects periodically

juancarlospaco, 2018-08-23:
# Reporting a problem?

No.

### Current Behavior

I can only select effects manually, 1 by 1, fixed.

### Your Environment

* Webcamoid information: 8.1.0,  x86_64.
* Operating System information: Arch Linux, rolling release, x86_64 64bit. 
* Any other useful information: I have the video loopback module installed.

# Want a new feature?

Make it possible to randomly use effects and change it automatically periodically.

I want an option to use one effect for 5 minutes, then other effect for 5 minutes, repeat.

Effect should be choosed randomly, a typical random function to choose one is Ok.

Effect duration on minutes should be configurable, an integer will do.

Dont worry about lost frames during effect switch, if video goes black for a blink its ok.

The new effect can be named `Random`, it only needs a positive integer for Minutes (periodic effect switch), if you want a list of effects to choose from, or else a random including all effects is Ok too.

Its a great way to use this App on Twitch, because the ease of use and the effects are cool, 
but you can only choose some manually and fixed, for a long stream random effects make it entertaining.

hipersayanX, 2018-08-23:
I think it somewhat related to #137, this falls under the category of transitions which are not implemented right now but it's a planned feature. My idea is to implement more complex layouts for presentations, screencasts, broadcast and so on.

juancarlospaco, 2019-01-16:
[Theres a ready made Lib, with C and C++ binding](https://www.youtube.com/watch?v=k1l3RdvwHeM) :slightly_smiling_face: