webcamoid/webcamoid #139: Unable to run Webcamoid on Linux Mint 19 Cinnamon (64bit)

Tuxman2, 2018-08-23:
Here are the messages I received from the terminal:

> QT5CT : Using QT5CT plugin
QT5CT : D-Bus System Tray : No
Segmentation error (Core Dumped)

Webcamoid version (from Ubuntu repositories) : Webcamoid 8.1.0+dfsg-7

Thanks for your help.

Note: Linux Mint 19 is based on Ubuntu 18.04.

hipersayanX, 2018-08-23:
Did tried qt5ct in Arch Linux and can't reproduce. This is the output in my case:

```
QT5CT : Using QT5CT plugin
QT5CT : D-Bus System Tray : Yes
```

Some things you can try:

1. Launch webcamoid from terminal as:

QT_QPA_PLATFORMTHEME=qtcurve webcamoid

You must have qtcurve installed.

2. Try running the [AppImage](https://github.com/webcamoid/webcamoid/releases).
3. Try purging qt5ct with _sudo apt purge qt5ct_ (or whatever equivalent package).
4. Maybe something from [here](https://forum.manjaro.org/t/solved-qt5ct-throws-segmentation-fault/48596) may help?

Tuxman2, 2018-08-24:
Hello,

First, thanks for your help. ;)

I tried all your solutions. Only the second works (AppImage file). With the first and the third solution I have still the Segmentation fault. 

Do you know if a flatpak file is available for Webcamoid ? By default, Linux Mint 19 use flatpak (for software not available in the repositories) but I can also use AppImage files.

The Webcamoid version from the Ubuntu repositories don't work. :(

Regards.

hipersayanX, 2018-08-24:
>Do you know if a flatpak file is available for Webcamoid?

No yet, but is on my plans to release a Flatpak and a Snap package. 

>The Webcamoid version from the Ubuntu repositories don't work. :(

Maybe the package needs a recompile, you can try [reporting it to them](https://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/), or maybe you can try [compiling](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-GNU-Linux) and debugging it.

Lvaskz, 2018-10-25:
[image]

Cool!!!!!!! Doyou have a date for this releases?

hipersayanX, 2018-10-27:
@Lvaskz for Webcamoud 8.5.0 probably between 1 or 2 months, for Flatpak and a Snap packages I haven't but for Webcamoid 9.0.0 for sure, not sure if I will made some middle releases.

hipersayanX, 2019-02-15:
Tried with LinuxMint 19.1 Cinnamon and can't reproduce the issue.

closed 2019-02-15 by hipersayanX

Calendis, 2020-04-23:
I know this is closed, but I have this issue as well
`qt5ct: using qt5ct plugin
qt5ct: D-Bus system tray: no
Segmentation fault (core dumped)`

hipersayanX, 2020-04-24:
@Calendis did you tried the [Daily Build](https://github.com/webcamoid/webcamoid/releases)?