webcamoid/webcamoid #141: Webcam settings not returned to normal after uninstall

karljpd, 2018-09-14:
### Summary

After installing Webcamoid, it changed the settings of my internal webcam (and external webcam). Uninstalling has had no effect. Was initially very dark and grainy so I reinstalled Webcamoid, changed the webcam settings to get it back to how it should be (as best I could by adjusting the brightness, saturation, sharpness, etc.). Uninstalled again, but now I have noticed that the webcam does not auto adjust for brightness. How can I get back to my original webcam settings? Why does uninstalling Webcamoid not remove all the changes?

I have tried uninstalling the internal webcam driver. Upon restart, the drivers are automatically reinstalled, but there is no change. Still has the webcamoid settings.

I think this person had a similar issue (the recommended action from the Microsoft agent had no effect):
https://answers.microsoft.com/en-us/windows/forum/windows_10-hardware/messed-up-webcam/d1cbcb98-f02e-4459-b484-7f8a2e02fb2b

For my external webcam, removing the device and then reinstalling returned it to its normal settings.

### Your Environment

* Webcamoid information: 8.1.0 64 bit
* Operating System information: Windows 10, Dell Inspiron 7658

hipersayanX, 2018-09-14:
>Why does uninstalling Webcamoid not remove all the changes?

Camera controls settings are stored in the camera hardware, installing, uninstalling or reinstalling the program or the driver will not take any effect.

>How can I get back to my original webcam settings? 

Normally when you disconnect the camera (in the case of external cameras) or turn off the computer (in the case of internal cameras), settings turns back to it's normals.

Also I must remove [this line](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/dshow/src/capturedshow.cpp#L986) that seems to be causing similar problems.

spiralstone, 2018-11-21:
I'm having the same issue.  My laptop webcam is unusable due to this issue.  I've uninstalled drivers, disabled the device, let the laptop run down the battery, shut off for a period of time, and it still comes back trashed.  Any other ideas or a fix for this is badly needed.  Thanks for your help...

hipersayanX, 2018-11-22:
@spiralstone 
Could [this](https://github.com/webcamoid/webcamoid/issues/148#issuecomment-435955819) fix the problem?
Do you remember if you did something special before (maybe moving some slider or something) or it just turned black as is?
Also, what hardware and OS version are you using?

Ok, there are a lot of people reporting the same and I'm getting very worried :worried:

hipersayanX, 2018-12-27:
This seems to be the same issue as #148, will continue there since it has more information.

closed 2018-12-27 by hipersayanX