webcamoid/webcamoid #142: Webcamoid Does Not Open

tuesdayholmes, 2018-09-16:
Hi all,

I'm running Linux Lite on my laptop and I am not too proficient with Debian.

When I try to open Webcamoid from the command line I get this output:

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Segmentation fault (core dumped)

But the program itself does not open. Am I missing some package?

ROBERT-MCDOWELL, 2018-09-17:
did you try as root?

hipersayanX, 2018-09-17:
>QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'

This message at least, indicate that you tried to run as root which is not ok, you must run it as an unprivileged (normal) user. If the segfault continues after that you can try using the [AppImage or any of the precompiled packages](https://github.com/webcamoid/webcamoid/releases).

As for the Segmentation fault, there has been a lot of people reporting the same in Debian and derivatives, which is unusual, @hpfn can you check this, maybe the package needs a recompile.

hpfn, 2018-09-17:
Hi,

It is ok in Debian Stable. But in Budgie (Ubuntu 18.04.01 - version 8.1.0+dfsg-7) no.
Reproducible builds are fine (version +dfsg-9) for more than a month:

https://tests.reproducible-builds.org/debian/rb-pkg/unstable/amd64/webcamoid.html

There is no '+b1' version - binNMU.
https://www.debian.org/doc/manuals/developers-reference/ch05.en.html#binary-only-nmu
https://www.debian.org/doc/manuals/developers-reference/ch05.en.html#nmu-binnmu

There are message about '/usr/lib/i386-linux-gnu/avkys/libMultiSrc.so' and 'libQt5'.
I can not copy and paste. The messages are not printed on terminal. They go to a
graphic window that do not allow that.

I do not know more right now.

Regards,
Herbert

tuesdayholmes, 2018-09-17:
Thanks all for your reply. 

Trying to run as root was a mistake but trying to run from standard permissions I get:

Segmentation fault (core dumped)

In the output. AppImage and other packages are also yielding no results.

hpfn, 2018-09-17:
Hi,

I found the file with the stacktrace in /var/crash/. Not much info I guess. 
It ends with:

Title: webcamoid crashed with SIGILL in MultiSrcElement::MultiSrcElement()
UpgradeStatus: Upgraded to bionic on 2018-05-02 (138 days ago)

hipersayanX, 2018-09-18:
@hpfn 
>There are message about '/usr/lib/i386-linux-gnu/avkys/libMultiSrc.so' and 'libQt5'.

Hmm..., right now I'm thinking that this could be related to #100, it shows the same behavior, not sure :confused: 

>I can not copy and paste. The messages are not printed on terminal. They go to a
graphic window that do not allow that.

Just take a capture. 

@tuesdayholmes do you have a _~/.config/Webcamoid_ file?

hpfn, 2018-09-18:
[_usr_bin_webcamoid.1000.txt](https://github.com/webcamoid/webcamoid/files/2392257/_usr_bin_webcamoid.1000.txt)

File attached!

hipersayanX, 2018-09-18:
```
Stacktrace:
 #0  0x0054b4e0 in ?? ()
 No symbol table info available.
 #1  0xacb89143 in MultiSrcElement::MultiSrcElement() () from /usr/lib/i386-linux-gnu/avkys/libMultiSrc.so
 No symbol table info available.
...
```

The problem is somewhere in the constructor but debug information is missing because **" No symbol table info available."**, can you compile with CONFIG+=debug?

hpfn, 2018-09-18:
[ubuntu_pks.txt](https://github.com/webcamoid/webcamoid/files/2394159/ubuntu_pks.txt)
[_usr_bin_webcamoid.1000.txt](https://github.com/webcamoid/webcamoid/files/2394160/_usr_bin_webcamoid.1000.txt)

If I have to install some dbgsym from Qt5 package let me know.

hipersayanX, 2018-09-19:
Sorry for the late response. 

>If I have to install some dbgsym from Qt5 package let me know.

No, that's more than enough, I can confirm is the same bug that #100.

```
Stacktrace:
 #0  0x005344e0 in qt_meta_stringdata_MediaSource ()
 No symbol table info available.
 #1  0xacb33bee in MultiSrcElement::codecLibUpdated (this=0x10ca3b0, codecLib=...) at multisrcelement.cpp:204
         state = AkElement::ElementStateNull
         media = {static null = {<No data fields>}, d = 0xbfaa26e8}
         loop = <optimized out>
         showLog = <optimized out>
 #2  0xacb34143 in MultiSrcElement::MultiSrcElement (this=<optimized out>) at multisrcelement.cpp:47
 No locals.
```

I did fixed that error in the master branch, but still can't figure out what's wrong with [that code](https://github.com/webcamoid/webcamoid/blob/28a7b62bc589c8f8452af70bd941156249ca34da/libAvKys/Plugins/MultiSrc/src/multisrcelement.cpp#L36) :confused:

ashl1, 2018-10-06:
Could I try to help in the case?

I use Kubuntu 18.04.1 LTS and with the latest updates webcamoid 8.1.0 Valgrind says:
```
==15306== Conditional jump or move depends on uninitialised value(s)
==15306==    at 0x7E9C5D7: __wmemchr_avx2 (memchr-avx2.S:260)
==15306==    by 0x7DF75D8: internal_fnwmatch (fnmatch_loop.c:168)
==15306==    by 0x7DFAA47: fnmatch@@GLIBC_2.2.5 (fnmatch.c:434)
==15306==    by 0x7E0EDE9: glob_in_dir (glob.c:1353)
==15306==    by 0x7E751B8: glob@GLIBC_2.2.5 (glob.c:1096)
==15306==    by 0x310A83BF: ??? (in /usr/lib/x86_64-linux-gnu/libv4l2.so.0.0.0)
==15306==    by 0x310A49A9: v4l2_fd_open (in /usr/lib/x86_64-linux-gnu/libv4l2.so.0.0.0)
==15306==    by 0x310A512C: v4l2_open (in /usr/lib/x86_64-linux-gnu/libv4l2.so.0.0.0)
==15306==    by 0x30E9006B: CaptureV4L2::updateDevices() (in /usr/lib/x86_64-linux-gnu/avkys/submodules/VideoCapture/libv4lutils.so)
==15306==    by 0x30E90D6B: CaptureV4L2::CaptureV4L2(QObject*) (in /usr/lib/x86_64-linux-gnu/avkys/submodules/VideoCapture/libv4lutils.so)
==15306==    by 0x30E8B0A1: Plugin::create(QString const&, QString const&) (in /usr/lib/x86_64-linux-gnu/avkys/submodules/VideoCapture/libv4lutils.so)
==15306==    by 0x4E66DEE: AkElement::loadSubModule(QString const&, QString const&) (in /usr/lib/x86_64-linux-gnu/libavkys.so.8.1.0)
==15306==  Uninitialised value was created by a stack allocation
==15306==    at 0x7DFA939: fnmatch@@GLIBC_2.2.5 (fnmatch.c:387)
==15306== 
==15306== Invalid read of size 2
==15306==    at 0x72DB053: QObject::event(QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5)
==15306==    by 0x2462AA47: MultiSrcElement::setState(AkElement::ElementState) (in /usr/lib/x86_64-linux-gnu/avkys/libMultiSrc.so)
==15306==    by 0x2462A31F: MultiSrcElement::codecLibUpdated(QString const&) (in /usr/lib/x86_64-linux-gnu/avkys/libMultiSrc.so)
==15306==    by 0x2462A811: MultiSrcElement::MultiSrcElement() (in /usr/lib/x86_64-linux-gnu/avkys/libMultiSrc.so)
==15306==    by 0x24629A5F: MultiSrc::create(QString const&, QString const&) (in /usr/lib/x86_64-linux-gnu/avkys/libMultiSrc.so)
==15306==    by 0x4E65509: AkElement::createPtr(QString const&, QString const&) (in /usr/lib/x86_64-linux-gnu/libavkys.so.8.1.0)
==15306==    by 0x4E65683: AkElement::create(QString const&, QString const&) (in /usr/lib/x86_64-linux-gnu/libavkys.so.8.1.0)
==15306==    by 0x1FA112CC: Pipeline::parse(QString const&) (in /usr/lib/x86_64-linux-gnu/avkys/libBin.so)
==15306==    by 0x1FA093FB: BinElement::setDescription(QString const&) (in /usr/lib/x86_64-linux-gnu/avkys/libBin.so)
==15306==    by 0x1FA158F4: ??? (in /usr/lib/x86_64-linux-gnu/avkys/libBin.so)
==15306==    by 0x1FA15A52: BinElement::qt_metacall(QMetaObject::Call, int, void**) (in /usr/lib/x86_64-linux-gnu/avkys/libBin.so)
==15306==    by 0x72B9C14: QMetaProperty::write(QObject*, QVariant const&) const (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5)
==15306==  Address 0x10 is not stack'd, malloc'd or (recently) free'd
```

Could you please write down commit in master that fixes the bahavior?
I'm trying to build webcamoid in Docker now. Do you have any Dockerfile for that?

hipersayanX, 2018-10-07:
@ashl1 when running valgrind you must compile the code as debug, otherwise it will not give you the line numbers, but even so my guess it is crashing [here](https://github.com/webcamoid/webcamoid/blob/28a7b62bc589c8f8452af70bd941156249ca34da/libAvKys/Plugins/MultiSrc/src/multisrcelement.cpp#L193), and valgrind is pointing at m_mediaSource because it's pointing to an invalid address, but m_mediaSource was allocated [here](https://github.com/webcamoid/webcamoid/blob/28a7b62bc589c8f8452af70bd941156249ca34da/libAvKys/Plugins/MultiSrc/src/multisrcelement.cpp#L36) and hold by the [shared pointer](https://github.com/webcamoid/webcamoid/blob/28a7b62bc589c8f8452af70bd941156249ca34da/libAvKys/Plugins/MultiSrc/src/multisrcelement.cpp#L29) and m_mediaSource was never touched again so it must still contains the reference to the object allocated in L29, but some how memory get corrupted there for no reason, we actually reached to that point in #100.

I blamed gcc and clang, but it seems it has nothing to do with the compiler, tried writing a small test case reproducing the behavior but no luck. There are [other plugins](https://github.com/webcamoid/webcamoid/blob/28a7b62bc589c8f8452af70bd941156249ca34da/libAvKys/Plugins/ACapsConvert/src/acapsconvertelement.cpp#L36) that follow the same scheme, but does not crash.

The idea with L36 was reducing the number of [checks](https://github.com/webcamoid/webcamoid/blob/93c4be6611e4db0cd4227afda37412398ae0bf93/libAvKys/Plugins/MultiSrc/src/multisrcelement.cpp#L71) (so those lines become like [this](https://github.com/webcamoid/webcamoid/blob/28a7b62bc589c8f8452af70bd941156249ca34da/libAvKys/Plugins/MultiSrc/src/multisrcelement.cpp#L57)) by providing a default dummy object, but well... you see the result.
So the only solution I've found was not creating the default object, but then we have the problem that this requires a massive rewrite of the code, so this can't be back ported and added as a simple patch to the package.

A complicated situation with a pesky bug.

>Could you please write down commit in master that fixes the bahavior?

It's already fixed in master, but patching in Debian and other systems with same problem is kinda complicated since it requires a massive rewrite, so the simplest solution I've found is just deleting _/usr/lib/avkys/libMultiSrc.so_ and disabling MultiSrc in build. At least as a palliative until we get to the next release.

>Do you have any Dockerfile for that?

The one used in [Travis](https://github.com/webcamoid/webcamoid/tree/master/ports/ci/travis).

hpfn, 2018-11-02:
Hi,

> Could you please write down commit in master that fixes the bahavior?
>
> It's already fixed in master, but patching in Debian and other systems with same problem is kinda complicated since it requires [...]

Debian will start the Freeze[0] period in Jan 2019.

[0] - https://release.debian.org/buster/freeze_policy.html

I would like to put the paliative solution in SID until
2018/12. But I did not find in raw build[1] instructions
how to do that.

[1] - https://github.com/webcamoid/webcamoid/wiki/Raw-build-and-install

How do I disable MultiSrc in build? Only remove libMultiSrc.so
file does not work.

Regards,
Herbert

hipersayanX, 2018-11-02:
>How do I disable MultiSrc in build? Only remove libMultiSrc.so

Remove [this line](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/Plugins.pro#L32), that will be enough.

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

closed 2019-01-22 by hipersayanX