webcamoid/webcamoid #143: Feature request: Image overlay

roncli, 2018-10-03:
Hey there!  I was wondering how easy it would be to add a plugin that simply overlays an image on top of the webcam's output.  It appears that the face detector already does something similar, but that follows the face.  I'm just looking for a static overlay where I can select a transparent PNG and have it be displayed on top of the camera's output.

Use cases would include having a static logo on the camera, displaying a "lower third" with information in it, etc.

hipersayanX, 2018-10-03:
Pretty much easier, a roughly code sketch is basically like this:

```
AkPacket ImageOverlayElement::iStream(const AkPacket &packet)
{
    QImage src = AkUtils::packetToImage(packet);
    QImage oFrame = src.convertToFormat(QImage::Format_ARGB32);

    QPainter painter;
    painter.begin(&oFrame);
    painter.drawImage(QRect(0, 0, 100, 100), QImage("your_image.png"));
    painter.end();

    AkPacket oPacket = AkUtils::imageToPacket(oFrame, packet);
    akSend(oPacket)
}
```

matheusgontijo, 2020-02-06:
That would be a great plugin!

xverges, 2020-02-28:
My only reason to install webcamoid was that I was expecting to be able to do this. Lots of my peers turn their camera off during videoconferencing, and I thought that it would be an interesting experiment to have an overlay on my video with some text reminding them of the effectiveness of video over voice. 

I then thought about other interesting possibilities, like make the team aware of my current mood, or sort-of caption myself for acronyms without having to switch to the chat window.

BTW, thanks for providing a virtual camera!

zerokol, 2020-08-17:
Did someone do that?