webcamoid/webcamoid #144: remove / uninstall

sentient, 2018-10-03:
How do I uinstall or remove webcamoid on linux? 

There is a wizard to install it. But now I'm lost where it got installed.

hipersayanX, 2018-10-03:
In the folder you installed Webcamoid (probably _~/webcamoid_), there is a binary called **WebcamoidMaintenanceTool**, you click and uninstall, also you can reset configs removing _~/.config/Webcamoid_ folder.

closed 2018-12-27 by hipersayanX

paul-nelson-baker, 2019-09-20:
Seems relevant to OSX users as well. The only difference is the default installation directory:
`~/Applications/webcamoid`

hipersayanX, 2019-09-21:
@paul-nelson-baker correct, if I remember well, the uninstall tool is installed there as **WebcamoidMaintenanceTool** as well, and it also appears in the launcher.

BoD, 2020-05-08:
Hello, I am a MacOS user, and after using the WebcamoidMaintenanceTool and rebooting, I still have Webcamoid, and would like to completely remove it.  Ideas?  I believe there should be some "kext" files somewhere, but where?

j256, 2020-05-08:
I found some reminants in my `~/Library` directory:

    cd ~/Library
    rm -rf Caches/Webcamoid
    rm -rf LaunchAgents/org.webcamoid.cmio.AkVCam.Assistant.plist
    rm -rf Preferences/com.webcamoid.*
    rm -rf Preferences/org.webcamoid.cmio.AkVCam.Assistant.plist
    rm -rf Saved\ Application\ State/com.webcamoidprj.webcamoid.savedState

hipersayanX, 2020-05-09:
@BoD the MaintenanceToo does not remove the virtual cameras yet, install Webcamoid again, remove the virtual cameras, and uninstall Webcamoid again.