webcamoid/webcamoid #146: How to use Virtual Camera?

Hexbugman213, 2018-10-20:
Hello! I am new to this software and I don't understand how to use the virtual webcam feature. I created one but I don't see it in any program I try it in. Don't ask, but I need to get the source from my desktop or a specific window into virtualbox.
I tested it with the windows camera app and Discord.
Specs: Windows 10 x64, ask if you need more

Hexbugman213, 2018-10-20:
I realized I have no idea what I"m doing. I'm going to read a bit more about this. sorry for bothering you.

closed 2018-10-20 by Hexbugman213

dmneufeld, 2019-08-24:
I have the same question:  How to make the virtual camera work? This was my main interest in installing Webcamoid: to use it for web-sites that require me to send streaming video (I'm teaching English online). Using the webcam built into my laptop gives me an upside-down video image. In Webcamoid, the image displayed is correct. But I don't know how to begin to configure it to send the video stream to my web-browser (Google Chrome or Vivaldi). Can you tell me how to set it up?

Operating system:  Feren OS
Computer:  Asus K52F laptop
Webcamoid version 8.6.1

hipersayanX, 2019-08-24:
@dmneufeld

First at all you must download and install either [akvcam](https://github.com/webcamoid/akvcam) or [v4l2loopback](https://github.com/umlaeute/v4l2loopback), if any of these come in the repositories of your distro just install from there. Then go to **General Options** and select the driver you want to use.

[image]

Create a new webcam and select it in **Devices**.

[image]

And then play whatever media you want to send (a webcam, the desktop, a video, etc.).

[image]

dmneufeld, 2019-08-28:
@hipersayanX  thanks for the response. Niether of the drivers were in my distro's repositories, so I installed akvcam according to the instructions here:
https://github.com/webcamoid/akvcam/wiki/Build-and-install

In the general options in webcamoid, there is still nothing showing up as an option for virtual camera driver in the general options. Maybe webcamoid cannot find where I have akvcam installed? Both akvcam and webcamoid have their own folders in my "home" directory.

Any ideas about how to get things set up correctly?

hipersayanX, 2019-08-31:
@dmneufeld 

>Niether of the drivers were in my distro's repositories

Which distro do you use?

>Both akvcam and webcamoid have their own folders in my "home" directory.

The driver does not install in /home, you must install it with DKMS, the module must be located in **/lib/modules/$(uname -r)/extra** as _akvcam.ko_ or _akvcam.ko.xz_.  

Also just in case, don't try to install the module unless you know how to compile and manage Linux modules, otherwise ask a maintainer to create a package for your distro, otherwise you will make a disaster with your system.

dmneufeld, 2019-09-02:
@hipersayanX
> Which distro do you use?

Feren OS

> Also just in case, don't try to install the module unless you know how to compile and manage Linux modules, otherwise ask a maintainer to create a package for your distro, otherwise you will make a disaster with your system.

Ok, I think I'll take that warning and uninstall what I've installed so far, and give up on this particular problem. Thanks for your help.

hipersayanX, 2019-09-02:
@dmneufeld
>Feren OS

You must install the **v4l2loopback-dkms** package.

    sudo apt-get install v4l2loopback-dkms

**NOTE**: Some programs [doesn't recognize the v4l2loopback driver](https://github.com/webcamoid/webcamoid/wiki/Virtual-camera-support#gnulinux) but overall is better than nothing.

>Ok, I think I'll take that warning and uninstall what I've installed so far, and give up on this particular problem. Thanks for your help.

Or you can ask a Debian/Ubuntu maintainer to create a package for akvcam, that would be really useful :smiley: