webcamoid/webcamoid #147: GIF Source displays incorrectly and in black and white.

tlbkwl1, 2018-10-24:
When creating a new source with a GIF, the file will appear and loop but with graphical errors (appears as a kind of diagonal split) and in black and white.  This occurs no matter what GIF is used.
[image]
Webcamoid version is Win32 8.1.0
OS is Win10 1803.

hipersayanX, 2018-10-25:
Can you upload the GIF file?

tlbkwl1, 2018-10-25:
[image]
Here is the GIF shown on the comment.
Below is another GIF I used which also showed the issue.
[image]

hipersayanX, 2018-10-25:
Thanks! will check.

hipersayanX, 2019-03-05:
Fixed in 8.6.0.

closed 2019-03-05 by hipersayanX