webcamoid/webcamoid #148: Webcam turns very dark, can't fix it

ProblemSolutionPlease, 2018-11-02:
Hello,
today I've installed your software, but my camera turned very dark. I tried restoring it by clicking button like "Restore settings", but this make the situation even worse. I uninstalled your software, reinstalled camera driver, reinstalled camera, restarted computer, turned off and on the computer, but nothing worked. I know that the settings are stored in my camera, but how to restore everything to normal. Please step-by-step manual.

OS: Windows 10
PC: HP ProBook
SW version: latest

Have a nice day

hipersayanX, 2018-11-02:
Does moving the controls manually changes something (Brightness, Contrast, Gain, etc.)?

ProblemSolutionPlease, 2018-11-03:
I think, that changing exposure makes the situation better, but I'd more like the solution, that will reset my cameras original settings.

hipersayanX, 2018-11-05:
Some cameras has a bad default quality, you can play with the sliders to make it a bit better.

[image]

Try, setting the controls with the ➡ to it's maximum.
Camera controls varies depending on the OS, vendor and camera.

spiralstone, 2018-11-23:
That does make the light better, but now it's like a filter is stuck on.  Changing filters does have an effect, but it seems like a constant filter is on and setting it to none does not reset back to a normal view.

hipersayanX, 2018-11-26:
@spiralstone can you upload a capture of the controls, like the one I posted above? also, do you have some effect activated?
[image]

JADzzz, 2018-12-06:
Same here! You broke my webcam, my camera turned very dark, can not fix it. I didn't try any effects. The first launch and my camera became so. How to solve it ???

hipersayanX, 2018-12-06:
Any chance to test the camera in GNU/Linux, so we can discard hardware problems? And try setting the controls with either Webcamoid or guvcview.

blogan86, 2018-12-21:
Add me to the list of people with a permanently messed-up webcam. I wish I had never installed this program! By playing with the controls I get to choose between too dark to see anything and completely washed out with light. Uninstalling webcamoid and resetting drivers does nothing. Hoping for a fix soon! There must be some way to programatically reset the camera hardware?

hipersayanX, 2018-12-22:
@blogan86 please, can you take an screenshot of the controls? also any chance to test the camera in GNU/Linux? **I need that information**.

blogan86, 2018-12-22:
The image was **not** ok before changing the controls. The very first time I installed webcamoid, the image was immediately much darker than the camera's factory default. Adjusting brightness/contrast/saturation/gamma can make the picture brighter, but only by creating a washed out image (like someone else said, it's like applying a filter).

[image]

I will look into testing it on Linux. I don't have it installed on this machine but maybe a live CD will work.

hipersayanX, 2018-12-22:
@blogan86 seriously thanks a lot and sorry! :+1:  I need a lot of feedback to solve this issue.

Some possible things I been thinking is that this is caused because the default values read from webcam were wrong. Can you take a picture of the controls after pressing the Reset button?
Also try setting the controls:

* Brightness, Contrast, Hue, Saturation, Sharpness: to the middle
* Gamma, White Balance, Exposure: more or less a quarter
* Color Enable: checked

>I will look into testing it on Linux. I don't have it installed on this machine but maybe a live CD will work.

Testing it in a LiveCD is more than enough, install is not necessary. You can download the [AppImage](https://github.com/webcamoid/webcamoid/releases). Try pressing the Reset button, so we can confirm it that fix the problem.

Keep me informed please.

hipersayanX, 2019-01-22:
This may be fixed in 8.5.0, but will leave this issue open for a while just in case.

thusithathilina, 2019-02-07:
I'm also facing the same issue. Dell latitude 7480. No solution works :(

benz0989, 2019-05-18:
this happened to me too.
my webcam just turned dark and i can't reset it back to normal.
i already tried removing everything but nothing works.
Has anyone found a fix for this problem yet.
im on a mac not pc

thusithathilina, 2019-05-18:
@benz0989 Give a try with following steps. It works for me

1. open webcamoid, reset the settings and close the webcamoid
2. open skype and go to webcam testing (then you probably able to see the cam was rest back to default settings)
3. close the skype and open the webcam through the default app

This works for me in Windows 10. But haven't tried on any other OS/versions

hipersayanX, 2019-05-18:
@benz0989 this is new, but since Mac and Windows use different API I don't think this has the same cause.

@ thusithathilina thanks, just in case, do you remember if you did something special to trigger the bug? did you tried reproducing the bug again?

Right now I need some way to reproduce the bug.
Also I don't think this is a hardware specific bug, my two suspects are DirectShow and Windows 10 camera permissions, not sure.

thusithathilina, 2019-05-19:
@hipersayanX  I can constantly reproduce the issue in Windows 10, Dell Latitude 7480 laptop.
Steps to reproduce

1. Open the webcam and change settings
2. Press the reset button
3. It does reset the values in the webcamoid, however, when I open the cam through windows app, effect is still there.

Hope this might help to figure out the reasons :)

tookitogo, 2019-08-27:
I can verify that this issue occurs on the Mac, too. The only fix I've found is using iGlasses (another utility) to reset control, and even so I'm not 100% convinced it's restored the original brightness completely. This affects both the internal FaceTime camera in a MacBook Air, as well as a separate Logitech C922. Note that in both cases, webcamoid does not show ANY camera controls whatsoever, just format, resolution, and frame rate. (And it doesn't show the C922's h.264 format, either.) It seems to me that webcamoid might not be correctly discovering the valid value range and creating the controls, but is nonetheless sending values out to the camera. Regardless, the fact that this survives to other applications, and that another third-party utility is needed to fix it, makes it a critical bug.

tookitogo, 2019-08-27:
> @tookitogo I think you should open a new issue specifically for Mac, since the capture framework is different for each operating system, so the problem can't be the same for two different operating systems.

Feel free to do that. I don't want to be the "owner" of the issue, since I am neither the only person to report it, nor the first! ;)

But yes, it is peculiar that the same issue is occurring on two different platforms.

> 
> > webcamoid does not show ANY camera controls whatsoever, just format, resolution, and frame rate. (And it doesn't show the C922's h.264 format, either.) It seems to me that webcamoid might not be correctly discovering the valid value range and creating the controls, but is nonetheless sending values out to the camera
> 
> Camera controls are [not implemented](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/avfoundation/src/captureavfoundation.mm#L129) in Mac, 

Relevant comment from link above:
>            // This controls will be not implemented since Apple doesn't
>            // provides an interface for controlling camera controls
>            // (ie. UVC controls).

Except that this is demonstrably untrue, since numerous first-party AND third-party utilities can and do offer camera controls. Logitech (just to list one example) has three different utilities, and [iGlasses](https://www.ecamm.com/mac/iglasses/) and [Mactaris Webcam Settings](http://mactaris.blogspot.com/p/webcam-settings.html) both support a wide array of cameras, including Apple's iSight/FaceTime cameras and numerous third-party models, with varying levels of control. (And no, I'm not talking about post-capture filters, I do mean actual camera control.)

>so it's absolutely impossible for Webcamoid to modify those values.
It wouldn't be a bad idea to change the user documentation to make clear what features are and are not supported on each platform. 

And above all, "impossible" or not, Webcamoid _is_ changing them.

tookitogo, 2019-08-27:
> @tookitogo the problem is that AVFoundation does not offer any official API to control the classic UVC controls: contrast, brightness, hue, saturation, gamma, and so on, and the controls offered by AVFoundation in the [documentation](https://developer.apple.com/documentation/avfoundation/avcapturedevice) (go to **Managing Exposure Settings** and bellow) does not seems to be the same, when I tried those controls with my webcam (Logitech C920) it did not worked. My guess is that iGlasses and Mactaris are probably making use of [this hack](https://phoboslab.org/log/2009/07/uvc-camera-control-for-mac-os-x).

Could be. Somebody else took that and expanded it into a library.

> > And above all, "impossible" or not, Webcamoid is changing them.
> 
> The problem is how. I need some way to reproduce the bug to understand what it could be happening, I need a step by step way to reproduce it.

For sure. In my case, merely switching around format settings (format/resolution/framerate) and switching sources (between internal FaceTime camera and C922) a few times was enough.

hipersayanX, 2019-09-07:
@tookitogo can you test the [latest build](https://bintray.com/webcamoid/webcamoid/webcamoid/daily#files) and tell me if it continues doing the same?, check that the webcam is ok before opening Webcamoid and then try triggering the bug again.

helpplease89, 2020-03-25:
Hi, 

I'm experiencing this same issue on Windows 10 (HP laptop camera), current build. Now, my webcam is unusable as it shows a dark picture regardless of software. 

I first encountered it on the latest 64 bit windows portable build. 
I've tried: 
- skype-related steps in this thread
- system restore 
- deleting all of the program files, stopping all program tasks running, and restarting. 
- uninstalling and reinstalling device via device manager
- installing the windows webcamoid install version and then uninstalling
- Removing any remaining .dll I can find that looks related

fuu1, 2020-03-25:
It seems to change Exposure and While Balance settings.
In addition it alters the resolution settings.

Skype seem to be able to find the right white balance settings:
[Settings image](https://i.imgur.com/kv8zITK.png)

hipersayanX, 2020-03-25:
I think it could be caused by some bug [here](https://github.com/webcamoid/webcamoid/blob/a0fd950b99822bdbc2158e8dc5c71b949f3e803f/libAvKys/Plugins/VideoCapture/src/dshow/src/capturedshow.cpp#L465), maybe some thing that is not initialized and is triggering setCameraControls() with false values, this make sense because some people informed this was caused just by running Webcamoid.
My second suspect would be something related to [Qml](https://github.com/webcamoid/webcamoid/blob/a0fd950b99822bdbc2158e8dc5c71b949f3e803f/libAvKys/Plugins/VideoCapture/share/qml/main.qml) but not sure about this.

Now this make clear for me that this bug is not "breaking" the webcam, but messing up with the controls, and this is something that IS fixeable.

gp0, 2020-04-04:
If you guys are looking for a way to reproduce this:

 (I don't know if this is cam/hardware specific)

I can reproduce this issue by running https://manycam.com/ in parallel (but letting webcamoid use the camera, manycam will just say "Can't start video"), right click on the preview in manycam -> properties -> "Camera Control". There's a checkbox for exposure "Auto", clicking it triggers the behaviour described (Auto on -> normal image, Auto off image too dark). The dialogue looks a lot like this: https://stackoverflow.com/questions/49603366/control-webcam-settings-programmatically-in-windows-10

Just thought a way to trigger the behaviour on demand might be helpful.

hipersayanX, 2020-04-07:
@gp0 it give me the Logitech settings dialog instead, but I've been seen that dialog before.

[image]

I've some idea of what could be the problem.

aaronbrighton, 2020-04-15:
I've been having the same issue as described here since installing Webcamoid and launching (didn't touch the configuration).  Camera white balance and brightness are completely off.

Tried uninstalling Webcamoid, the Webcam driver, etc...  Camera settings are now stuck on whatever Webcamoid set them too.

For the the time being I've tried adjusting the settings in Webcamoid to make the best of things, but I still get comments from people mentioning that it looks like I have a filter applied to my camera or that my video is dark.  Contemplating buying an external webcam to remedy the issue, crazy right?

Has anyone found a way to reset the settings back to original (on Windows), either through a third party application or manually through the same API webcamoid uses -- or are the original camera settings now officially lost due to this bug?  I assume the only way to get the settings back would be to try to pull the settings from whatever API on a matching computer configuration that hasn't been affected by this bug?

Windows 10 (latest updates)
Dell XPS 15 9570

thusithathilina, 2020-04-15:
@aaronbrighton Have you tried the steps that I've mentioned with the use of skype?

aaronbrighton, 2020-04-15:
> 
> 
> @aaronbrighton Have you tried the steps that I've mentioned with the use of skype? It doesn't restore the originals, but I was able to get somewhat good results. Mine is Dell Latitude E7480

That worked!  You're amazing, thanks.  I assume it didn't revert the settings, but alternatively applied the correct adjustments to the camera settings for an ideal white balance / exposure / brightness configuration as you mentioned.

For anyone else needing more detailed instructions, use the consumer version of Skype and open up your video settings:

1. Check off automatic white balance, press Apply.
2. Check off automatic exposure (second tab), press Apply.
3. Exit Skype (if you so desire), but don't let it "forget" your user settings.

The changes have persisted into other applications that use the webcam.

I also uninstalled Skype and the new settings seem to have persisted, so yay!

Olafski, 2020-04-15:
Just want to add this happened to me as well on a Dell Latitude. Quite the scare. With this many people having this issue, you may want to post a warning on the download page to be honest.

Thankfully the "Skype fix" seems to have worked.

gp0, 2020-04-15:
@hipersayanX if you put out a build (i checked https://bintray.com/webcamoid/webcamoid/webcamoid/daily, seems the last one is from January 28th?) i'll glady test and report back if your commit fixed the issue.

hipersayanX, 2020-04-16:
@Olafski I did not added any warning because the number of people affected by this bug was too low compared to the number of downloads, the best I was able to do was pinning up this issue so I can get any clue of what was happening.

@gp0 I've to fix the build, I'll do in few hours later in the morning, I'll advice when ready.

In theory this bug may be fixed in the last commit, after connecting the comments from @blogan86 @fuu1 and @gp0 and the Logitech setting, I come to the conclusion that I was missing the **Auto** controls that were controlled by the [pCapsFlags](https://docs.microsoft.com/en-us/previous-versions/ms784399%28v%3dvs.85%29) argument, Microsoft documentation is ambiguous explaining what I'm supposed to do with those flags.
When the user started Webcamoid in Windows, it sets the wrong values for auto controls, and that's what could be causing this bug. And I'm sure that this was the common point between all the reports.

hipersayanX, 2020-04-17:
@gp0 @Olafski I've uploaded the [daily builds](https://github.com/webcamoid/webcamoid/releases/tag/untagged-d3a4aa8a9bb5aec0c4c5) to Github Releases, isn't sorted in folders like Bintray but is something at least.

gp0, 2020-04-18:
@hipersayanX :tada: Looks like that fixed it! Can report it's working for me. Started up the daily build (portable win64) a couple of times, worked from the get go. Thanks!

If it's not fixed on start up, checking "Exposure (Auto)" in the source settings should do the trick for the people stuck on that.

thusithathilina, 2020-04-22:
I also tested the latest release on Win 10 x64 on a Dell Latitude E7480. Now the issue seems to be fixed :) Thanks

benbull, 2020-04-26:
I'm also having this issue, tried the daily build as recommended above but the colour, white balance and saturation are all off.  I can't seem to enable auto white balance in Skype anymore.  Would appreciate some help to get my camera back to the default settings before trying Webamoid

ayushjain1144, 2020-05-01:
The same issue happened with me on ubuntu 16.04. 

I installed v4l2cup software and pressed reset on all settings. It solved my issue.

hipersayanX, 2020-05-01:
@ayushjain1144 but you are using the latest version or an old one?
Also this issue only applies to Windows, as the problem is clearly identified in this platform, other platforms need their own issue.

ayushjain1144, 2020-05-01:
@hipersayanX I installed using apt-get. Not sure if it had the latest version.

hipersayanX, 2020-05-02:
@ayushjain1144 pretty sure you have not, latest stable release is [8.7.1](https://github.com/webcamoid/webcamoid/releases), and Ubuntu 16.04 has 6.2.0 in their repositories that's too old and a lot of bug fix has been made since then, test latest version or daily build instead.

thusithathilina, 2020-05-07:
Unfortunately, it seems the issue still exists. 
Steps to reproduce (Windows 10, Dell Latitude 7480)
1. Change brightness
2. Disable auto exposure and change exposure to make the camera look dark
3. Now try reset settings
4. Setting shown to be rest, however, the actual look not get restored.
I had to use the skype trick to restore back

hipersayanX, 2020-05-08:
@thusithathilina do 1, 2, 3 steps again and show me and screenshot of the controls.

thusithathilina, 2020-05-09:
> @thusithathilina do 1, 2, 3 steps again and show me and screenshot of the controls.
> I've some idea what it could be, at least one of the problems seems to be solved.

[image]

hipersayanX, 2020-05-09:
@thusithathilina in that state, there are some thing I want you to do

1. Turn off and then turn on the (Auto) controls, does it change the picture?
2. Try moving the sliders (or the spinboxs) just a bit (left or right), does it change the picture?
3. Give me an screenshot of the Skype controls.

Always put the camera in the black screen state before each step, so we can have a point of comparison.
Also, you are using an old daily build, download the latest one.

thusithathilina, 2020-05-09:
@hipersayanX do you want me to follow the steps with the current version or with the latest version?

JamesWirth1, 2020-05-09:
It probably is an exposure issue.  Check mark it and see if it fixes it.

hipersayanX, 2020-05-11:
@thusithathilina with the [daily build](https://bintray.com/webcamoid/webcamoid/webcamoid/daily) (I've finally fixed it).

thusithathilina, 2020-05-11:
@hipersayanX I can't see any settings button in the latest daily build (at least in the win64 portable version)

hipersayanX, 2020-05-17:
@thusithathilina Other people reported the same behavior in Linux too, but I can't reproduce it, I've made some changes that could solve that problem, but since I can't reproduce the issue that's not guaranteed, can you test the latest daily build again?

thusithathilina, 2020-05-17:
@hipersayanX Yep, I can still reproduce the issue on WIn 10. 
Steps
1. Just turn off the Auto exposure (My cam get total black when doing that)
2. Try Reset (It will not come back to normal)

hipersayanX, 2020-05-18:
@thusithathilina 

>. Try Reset (It will not come back to normal)

It must return to be checked by default, I think the problem is related to Qml.

thusithathilina, 2020-05-18:
@hipersayanX What I meant was the settings get back to normal (or at least it looks normal as the Auto exposure slider set back to turn on) However, the real look does not get reset.

JohemianKnapsody, 2020-05-26:
Also had this issue after using a portable 64 bit Windows version. My webcam in all other programs became very dark, and after reopening Webcamoid and clicking "Reset", the preview cycles between the very dark, greyscale preview and a brighter version with a strong green tint. Manually changing the settings eventually got it back to an acceptable preview, but I'm sure it's not as good as before I used Webcamoid.

If the program is adjusting settings that will affect all other programs, I'm assuming that a different program that also affects the settings will be able to automatically fix it. Does anyone know of another free program which correctly resets these settings for all other programs?

aaronbrighton, 2020-05-26:
> Does anyone know of another free program which correctly resets these settings for all other programs?

@JohemianKnapsody I was able to use Skype

https://github.com/webcamoid/webcamoid/issues/148#issuecomment-613778881

litomd, 2020-07-10:
Windows 10 hack: opening Windows Camera App works too. Open Camera App, check image is fine, close app. It restores camera settings.

samliddicott, 2020-07-20:
I got very dark camera view on windows 10 with Intel SR300 camera.

In webcamoid I could improve this by increasing exposure slider from -6 to -3 (any higher than -3 causes motion blur) and increasing gain from 64 to 100 and a small increase on gamma.

I'm surprised that it affects other uses of the same camera outside of webcamoid.

Using Intel Realsense Viewer I could re-enable auto-exposure

hipersayanX, 2020-07-22:
Just in case, this problem was more or less solved in 1cf5159 and the fix is available in the daily build. What I did not tried yet is if the exposure can be changed when the automatic exposure controls is checked, in theory it should not be like that.

@samliddicott 

>I'm surprised that it affects other uses of the same camera outside of webcamoid.

Because brightness, contrast, gamma and all other controls are applied at hardware level, not local to the application, when you change the brightness in one app, it's modified in all apps, and when you disconnect the webcam the controls returns to its factory settings.

samliddicott, 2020-07-23:
The daily build solves this problem by means of a pre-ticked auto-exposure box (although the initial view is still dark)

Thanks for the fix!

VA1DER, 2021-01-12:
This issue definitely still exits in the latest release.  Whatever way the release version uses to turn off the auto-exposure does it in a way that some other software can't turn back on.  VirtualDub2 and OBS both couldn't fix the problem.

A pinned issue in the github tracker is **not sufficient warning** to people that if they use this software it may permanently affect their webcam in a way only finding a daily build will fix.  A pinned issue in the github tracker is like a notice in the bottom of a locked file cabinet in the back of a disused lavatory with a sign on the door saying "beware of the leopard".

I'm trying to cut a line between seriousness and a bit of humour (with apologies to Douglas Adams) because, really, what needs to be gently pointed out here is that keeping your release version with its current behaviour is irresponsible.  You seriously need a warning, need to rebuild the release version with this fixed, or to make daily the new release.

edtkw, 2021-01-12:
Is there anyway to have a button click or something to reset it? I opened the portable version on a windows laptop that I do not have the administrative rights to. So I cant reinstall the webcam.

[11 more replies]

commit 1cf51596 (2020-04-15): Handle image and camera controls flags properly, hopefully this may fix
#148 once for all.
closed 2021-02-16 by hipersayanX