webcamoid/webcamoid #150: Settings doesn't save when exit

Skornflex, 2018-11-05:
Hi,
Got a little problem guys, i have to set a speechicam 7 in a school, but the original software is not supported on Ubuntu, that's why i've choose webcamoid, all works great (nice job), nice resolution, good FPS, etc. But, when i exit webcamoid, all settings are reset, i set RGB3, 2048x1536, with contrast at 5, default is MJPG at 1600x1200 witch contrast at 0 (this default settings lags a lot for me).
The problem is that teachers "won't be able" to do that each time if you know what i mean.
So is this a bug, or it's just not possible to change default settings ?

PS: Sorry for my bad English, i'm from France, don't hesitate to ask me back if you've not understand something.

hipersayanX, 2018-11-05:
This is related to #62, I was supposed to implement this for [8.5.0](https://github.com/webcamoid/webcamoid/wiki/Development-Roadmap#storingrestoring-device-configs) but will most probably make a 8.6.0 release to implement this and some other small issues I left there.

hipersayanX, 2018-12-27:
Duplicate of #62.

closed 2018-12-27 by hipersayanX