webcamoid/webcamoid #155: devel-version on linux: missing new QT5.10++ QtQuick Controls 2 required

mike-hb, 2019-01-18:
### Summary

today's devel version from git does not compile on opensuse Thumbleweed (qtcreator)

### Current Behavior

Error in Makefile line 46

started in the QT creator (after the prep steps mentioned in the readme.md)

### Suggestions and tips

As the QT5 interface changed in the past, there seems to be a missing adaption to this new QT5.10++
May be, I am wrong with my suggestions...

Thanks in advance

hipersayanX, 2019-01-19:
It actually [compiles](https://travis-ci.org/webcamoid/webcamoid/jobs/481485682#L2573) in [Tumbleweed](https://travis-ci.org/webcamoid/webcamoid/jobs/481485682#L435).
Check you have your repositories updated and all [dependencies](https://github.com/webcamoid/webcamoid/blob/4e3287a91bf48bcca1ecdff6d3b047d692e09944/ports/ci/travis/install_deps.sh#L174) installed.

mike-hb, 2019-01-19:
Thanks for the fast reply -- I think the problem has been not in the code but in front of the keyboard  ;-)
Yes, it compiles! -- And it works!

But I got an error message with the Makefile again missing rules for install with deploy due to possible missinterpreted your instructions for this part -- see the attached part of the compile log.
[partly-log-of-compile.txt](https://github.com/webcamoid/webcamoid/files/2776016/partly-log-of-compile.txt)
I am looking forward for the release of the Version 8.5!
Thanks for now
Michael

hipersayanX, 2019-01-19:
You probably have enabled the shadow build in Qt Creator, right? if that's the case you must set the environment variable **BUILD_PATH** to wherever is located the shadow build, for example:

```
export BUILD_PATH=~/webcamoid-shadow
python3 ports/deploy/deploy.py
```
or simply

```
BUILD_PATH=~/webcamoid-shadow python3 ports/deploy/deploy.py
```

Implying that the shadow build is located at that folder.

hipersayanX, 2019-01-22:
Issue seems to be solved, closing.

closed 2019-01-22 by hipersayanX