webcamoid/webcamoid #158: [regression] 8.5.0 crashes on FreeBSD

yurivict, 2019-01-23:
```
$ webcamoid 
QKqueueFileSystemWatcherEngine::addPaths: kevent: Invalid argument
Segmentation fault
```

```8.1.0``` works fine.

yurivict, 2019-01-23:
```
#1  0x00000008044ecf86 in __dynamic_cast () at /usr/local/lib/gcc8/libstdc++.so.6
#2  0x000000080456bdb0 in bool std::has_facet<std::ctype<char> >(std::locale const&) () at /usr/local/lib/gcc8/libstdc++.so.6
#3  0x000000080455f154 in std::basic_ios<char, std::char_traits<char> >::_M_cache_locale(std::locale const&) () at /usr/local/lib/gcc8/libstdc++.so.6
#4  0x000000080455f5d0 in std::basic_ios<char, std::char_traits<char> >::init(std::basic_streambuf<char, std::char_traits<char> >*) () at /usr/local/lib/gcc8/libstdc++.so.6
#5  0x0000000804500953 in std::ios_base::Init::Init() () at /usr/local/lib/gcc8/libstdc++.so.6
#6  0x000000082a8b4c56 in __static_initialization_and_destruction_0(int, int) (__initialize_p=1, __priority=65535) at /usr/local/lib/gcc8/include/c++/iostream:74
#7  0x000000082a8b4c89 in _GLOBAL__sub_I_ipcbridge.cpp(void) () at src/ipcbridge.cpp:3411
#8  0x000000080073a7ab in objlist_call_init (list=<optimized out>, lockstate=<optimized out>) at /usr/src/libexec/rtld-elf/rtld.c:2657
#9  0x000000080073f009 in dlopen_object (name=0x800885000 "z\270", <incomplete sequence \325>, fd=<optimized out>, refobj=<optimized out>, lo_flags=10, mode=1, lockstate=0x8
00000002) at /usr/src/libexec/rtld-elf/rtld.c:3356
#10 0x000000080073b836 in rtld_dlopen (name=0x81a86d898 "/usr/local/lib/avkys/libVirtualCamera.so", fd=-1, mode=<optimized out>) at /usr/src/libexec/rtld-elf/rtld.c:3231
#11 0x000000080386a7e5 in  () at /usr/local/lib/qt5/libQt5Core.so.5
#12 0x00000008038654ea in  () at /usr/local/lib/qt5/libQt5Core.so.5
#13 0x0000000803865913 in  () at /usr/local/lib/qt5/libQt5Core.so.5
#14 0x00000008009923b2 in AkElement::createPtr(QString const&, QString const&) (pluginId=..., elementName=...) at src/akelement.cpp:260
#15 0x0000000800992311 in AkElement::create(QString const&, QString const&) (pluginId=..., elementName=...) at src/akelement.cpp:243
#16 0x0000000000416753 in MediaTools::MediaTools(QObject*) (this=0x7fffffffe770, parent=0x0) at src/mediatools.cpp:95
#17 0x0000000000415c00 in main(int, char**) (argc=1, argv=0x7fffffffe7f8) at src/main.cpp:89
(gdb) 
```

yurivict, 2019-01-23:
The above crash is when built with gcc-8. This port was switched to gcc because of crashes when built with clang.

Now 8.5.0 doesn't crash when built with clang, it just doesn't show the video.
The debug version of 8.5.0 built with clang seems to work fine.

hipersayanX, 2019-01-24:
The QFileSystemWatcher bug was not fixed yet? :confused: 

>it just doesn't show the video at all.

How is that? it does not show neither the webcam or the desktop?

>The debug version of 8.5.0 built with clang seems to work fine.

Will check, only tried with debug.

yurivict, 2019-01-24:
> How is that? it does not show neither the webcam or the desktop?

It opens with a gray screen.

yurivict, 2019-02-03:
Ping. How can I solve this?

hipersayanX, 2019-02-04:
The crash or the gray screen?

For the crash, comment [these](https://github.com/webcamoid/webcamoid/blob/d0746abeb19f50f51443c1a34ebbbddf0b082de1/libAvKys/Plugins/VirtualCamera/src/v4l2sys/VCamIPC/src/ipcbridge.cpp#L2113) [lines](https://github.com/webcamoid/webcamoid/blob/d0746abeb19f50f51443c1a34ebbbddf0b082de1/libAvKys/Plugins/VideoCapture/src/v4l2sys/src/capturev4l2.cpp#L1311), but a much better solution would be fixing Qt packages and reporting the bug to [Qt developers](https://bugreports.qt.io/secure/Dashboard.jspa).

For the gray screen, trace the video packets from [here](https://github.com/webcamoid/webcamoid/blob/d0746abeb19f50f51443c1a34ebbbddf0b082de1/libAvKys/Plugins/VideoCapture/src/v4l2sys/src/capturev4l2.cpp#L360) to [here](https://github.com/webcamoid/webcamoid/blob/d0746abeb19f50f51443c1a34ebbbddf0b082de1/StandAlone/src/videodisplay.cpp#L52) and try to figure out where is the bug, it could be some missing rendering flag [like](https://github.com/webcamoid/webcamoid/blob/d0746abeb19f50f51443c1a34ebbbddf0b082de1/StandAlone/src/main.cpp#L77), it could be anything.

hipersayanX, 2019-02-09:
@yurivict any chance to fix this bug in Qt packages? I will prefer that rater than adding more patches to Webcamoid.

yurivict, 2019-02-09:
I'll try.

hipersayanX, 2019-02-25:
@yurivict any advance with this?

hipersayanX, 2019-02-26:
Can't reproduce neither the crash or or the gray screen in TrueOS (FreeBSD 12.0), only tested with clang, but disabled anything related to QFileSystemWatcher::addPaths preventively.

hipersayanX, 2019-03-05:
This is maybe solved in 8.6.0 but I'm not able to test it. But again I think that this bug must be solved in FrreBSD/Qt, not Webcamoid.

yurivict, 2019-03-06:
gcc-built version 8.6.0 crashes in static constructors:
```
[New LWP 101369 of process 96186]
[New LWP 102163 of process 96186]
[New LWP 103169 of process 96186]
QKqueueFileSystemWatcherEngine::addPaths: kevent: Invalid argument
[New LWP 103357 of process 96186]
[New LWP 103371 of process 96186]
[New LWP 103380 of process 96186]

Thread 1 received signal SIGSEGV, Segmentation fault.
0x00000008056fed78 in vtable for __cxxabiv1::__si_class_type_info () from /lib/libcxxrt.so.1
(gdb) bt
#0  0x00000008056fed78 in vtable for __cxxabiv1::__si_class_type_info () at /lib/libcxxrt.so.1
#1  0x00000008044ecf86 in __dynamic_cast () at /usr/local/lib/gcc8/libstdc++.so.6
#2  0x000000080456bdb0 in bool std::has_facet<std::ctype<char> >(std::locale const&) () at /usr/local/lib/gcc8/libstdc++.so.6
#3  0x000000080455f154 in std::basic_ios<char, std::char_traits<char> >::_M_cache_locale(std::locale const&) () at /usr/local/lib/gcc8/libstdc++.so.6
#4  0x000000080455f5d0 in std::basic_ios<char, std::char_traits<char> >::init(std::basic_streambuf<char, std::char_traits<char> >*) ()
    at /usr/local/lib/gcc8/libstdc++.so.6
#5  0x0000000804500953 in std::ios_base::Init::Init() () at /usr/local/lib/gcc8/libstdc++.so.6
#6  0x000000082b6a8c56 in __static_initialization_and_destruction_0(int, int) (__initialize_p=1, __priority=65535) at /usr/local/lib/gcc8/include/c++/iostream:74
#7  0x000000082b6a8c89 in _GLOBAL__sub_I_ipcbridge.cpp(void) () at src/ipcbridge.cpp:3411
#8  0x000000080073a7ab in objlist_call_init (list=<optimized out>, lockstate=<optimized out>) at /usr/src/libexec/rtld-elf/rtld.c:2657
#9  0x000000080073f009 in dlopen_object
    (name=0x800885800 "z\270", <incomplete sequence \325>, fd=<optimized out>, refobj=<optimized out>, lo_flags=10, mode=1, lockstate=0x7fff00000002)
    at /usr/src/libexec/rtld-elf/rtld.c:3356
#10 0x000000080073b836 in rtld_dlopen (name=0x81aa6ba98 "/usr/local/lib/avkys/libVirtualCamera.so", fd=-1, mode=<optimized out>) at /usr/src/libexec/rtld-elf/rtld.c:3231
#11 0x000000080386a7e5 in  () at /usr/local/lib/qt5/libQt5Core.so.5
#12 0x00000008038654ea in  () at /usr/local/lib/qt5/libQt5Core.so.5
#13 0x0000000803865913 in  () at /usr/local/lib/qt5/libQt5Core.so.5
#14 0x00000008009923b2 in AkElement::createPtr(QString const&, QString const&) (pluginId=..., elementName=...) at src/akelement.cpp:260
#15 0x0000000800992311 in AkElement::create(QString const&, QString const&) (pluginId=..., elementName=...) at src/akelement.cpp:243
#16 0x0000000000416753 in MediaTools::MediaTools(QObject*) (this=0x7fffffffe780, parent=0x0) at src/mediatools.cpp:95
#17 0x0000000000415c00 in main(int, char**) (argc=1, argv=0x7fffffffe808) at src/main.cpp:89
```

Do you use static constructors?

yurivict, 2019-03-06:
clang-built version 8.6.0 works fine now.

hipersayanX, 2019-03-06:
>Do you use static constructors?

If you mean static global objects, [yes a lot](https://github.com/webcamoid/webcamoid/blob/f1678d15a59d902a14bda22062ee5d3639a954d7/libAvKys/Plugins/VirtualCamera/src/v4l2sys/VCamIPC/src/ipcbridge.cpp#L290), but that error is pointing [here](https://github.com/webcamoid/webcamoid/blob/f1678d15a59d902a14bda22062ee5d3639a954d7/libAvKys/Plugins/VirtualCamera/src/v4l2sys/VCamIPC/src/ipcbridge.cpp#L3411) and that is just constructing a QString on fly :confused: . Maybe it it need to be more explicit with QString {...} or QString(...) not sure.

>clang-built version 8.6.0 works fine now.

Forget GCC and go with Clang.

yurivict, 2019-03-06:
> Forget GCC and go with Clang.

Well, this indicates that your use of global objects with statically called constructors is invalid, and even clang build can break any time, when the order of their invocation will change in a certain way.

hipersayanX, 2019-03-07:
How would you rewrite it?

yurivict, 2019-03-07:
Now it looks to me that this is possibly a bug in the gcc compiler on FreeBSD. I created a ticket for people responsible for gcc on FreeBSD: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=236344

closed 2019-03-07 by yurivict

hipersayanX, 2019-03-07:
Anyway, the VirtualCamera module is not that much important in FreeBSD since there is not a virtual camera driver yet. You can disable it with:

```
qmake Webcamoid.pro NOVCAM=1
```