webcamoid/webcamoid #159: Virtual Camera Enabled on MacOS Eats Up Memory

graywolf336, 2019-01-25:
### Summary

After manually launching webcamoid on MacOS 10.14.2 (see [here](https://github.com/webcamoid/webcamoid/issues/156#issuecomment-457572936)), I am able to input my rtsp camera and all is fine. Then I enable the Virtual Camera and all is good for a few minutes. However, after leaving webcamoid up and running for a while my system complains about running out of memory and webcamoid is taking up 32gb+ of RAM (no clue how it got this high when I only have 16gb).

### Current Behavior

Seems like the virutal camera has a memory leak because as soon as I enable it the memory usage jumps up 1gb every four seconds.

### Expected Behavior

Enabling virutal camera without consuming all of my RAM.

### Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1. Open webcamoid
2. Add a rtsp source
3. Create a new virtual camera
4. Enable the virtual camera
5. Watch the memory usage skyrocket

### Your Environment

* Webcamoid information: 8.5.0 64bits
* Operating System information: MacOS 10.14.2, 16gb ram, 3.4GHz Intel Core i5
* Any other useful information: Logs: https://pastebin.com/pzzZAZxh

graywolf336, 2019-02-08:
Just changed the description as it happens with any source and not just an rtsp source. I can use it for about 30 seconds before everything crawls to a halt and it gets all distorted due to eating up the memory. I know it is the virtual camera because when I start webcamoid with activity monitor up (mac's task manager; I can see the memory usage) and enable the the virtual camera, the memory skyrockets. However, when I disable the virtual camera everything returns back to normal.

Happens with the daily build as of this comment (February 8th, 2019 at 2:48pm CST).

hipersayanX, 2019-02-09:
Will check it's probably something related to the plugin. Also, which target program?

graywolf336, 2019-02-09:
Not sure I know what you mean by "which target program?"... I have a webcam input that then goes to a new virtual camera and as soon as I enable the virtual camera the memory usage starts to sky rocket

hipersayanX, 2019-02-09:
>Not sure I know what you mean by "which target program?"

The program that is receiving the frames, Chrome, Firefox for example, if there is not a target program then it could be more likely a problem in the IpcBridge, anyway will check.

graywolf336, 2019-02-09:
Yeah, nothing was using it when it starts.

hipersayanX, 2019-02-14:
This issue was fixed in the daily build.

graywolf336, 2019-02-14:
I will test it tomorrow, thanks!

graywolf336, 2019-02-18:
Verified it has been fixed in the daily builds. :) thanks!!!

hipersayanX, 2019-03-05:
May be fixed in 8.6.0.

closed 2019-03-05 by hipersayanX