webcamoid/webcamoid #161: Project ERROR: Library 'libatomic' is not defined.

hpfn-d, 2019-02-09:
Hi,

Webcamoid 8.5.0 passed Debian CI tests 3 times. Then
failed today.

The error message is:
Project ERROR: Library 'libatomic' is not defined.
make[1]: *** [Makefile:47: sub-Lib-make_first-ordered] Error 3
make[1]: Leaving directory '/tmp/autopkgtest-lxc.oodkc_cc/downtmp/build.7YV/src/libAvKys'
make: *** [Makefile:46: sub-libAvKys-make_first-ordered] Error 2
dh_auto_build: make -j2 returned exit code 2

It was used different sources to build. As the message says 'not defined'
I tried to disable libatomic (without success). I would like to confirm that 
is not a Webcamoid problem.

Please take a look at: [Debian CI tests](https://ci.debian.net/packages/w/webcamoid/testing/amd64/).

Regards,
Herbert

hipersayanX, 2019-02-09:
Webcamoid [builds ok](https://travis-ci.org/webcamoid/webcamoid/jobs/490995774#L438) in Debian testing.

```
grep -r libatomic
```

Return no results, which means it's not being used anywhere in the project, so is more likely a problem with your build environment.

hpfn-d, 2019-02-09:
On 2/9/19 2:50 PM, hipersayanX wrote:
> Webcamoid builds ok <https://travis-ci.org/webcamoid/webcamoid/jobs/490995774#L438> in Debian testing.
> 
> |grep -r libatomic |
> 
> Return no results, which means it's not being used anywhere in the project, so is more likely a problem with your build environment.
> 

I also did the same grep.

Thanks for Travis!
Herbert

hipersayanX, 2019-02-09:
>Thanks for Travis!

And the person that that recommend me to use it (@techtonik) :smile:

closed 2019-02-09 by hipersayanX