webcamoid/webcamoid #162: Using Webcamoid libuvc functions

Tilix4, 2019-02-15:
Hi, I'm very impressed by this software ! At what knocks me out, it's that you've find a way to use libuvc without ruining windows' drivers. How did you achieve that ? I'm trying to implement it in a project with opencv but it's not working and the only way I found is to use webcamoid aside, but it's not very handy.

I'm encountering this problem : https://github.com/ktossell/libuvc/issues/12#issuecomment-109901302
I cannot handle the webcam through libuvc without using Zadig. And you've succeeded it.

Is there a way for me to use your method ? I've read your code but I don't figure out how you did it. (I'm not a c++ pro...).

Thank for this software in any case !

hipersayanX, 2019-02-15:
In Windows, Webcamoid make use of DirectShow for webcam capture, and alternatively Media Foundation if you compile it with MSVC. I was not able to test libuvc in Windows since there are not any pre-compiled package yet, an I need at least one of these options to be available:

- Precompiled libraries for libusb an libuvc.
- A [MSYS2](https://github.com/msys2) package.
- An [Arch MinGW](https://aur.archlinux.org/packages/?O=0&K=mingw-w64-) package.

Otherwise it will be kinda hard integrating into the build pipeline (either Travis, AppVeyor or manually).

Tilix4, 2019-02-16:
Okay thank you.
But, something I don't get, is how you achieved to shunt windows drivers to have control and streaming at the same time. I have the webcam video stream with all applications but I get `device not found: -5` with the `uvc_open()` command.
If I want to have control, I have to go with zadig, and I can access the device through libuvc, but I loose video stream for every application.

hipersayanX, 2019-02-17:
>how you achieved to shunt windows drivers to have control and streaming at the same time.

Dou you mean how the program capture from the webcam and at same time it send the frames to the virtual camera? If that the case, in DirectShow each webcam is mapped as an element, a plugin that is loaded in the target program, here is the [documentation](https://docs.microsoft.com/en-us/windows/desktop/directshow/writing-directshow-filters) for writing a DirectShow plugin, [here](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCamera/src) you have the source code of the plugin, and [this](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCamera/src/plugin.cpp) is the "main" file.
You must define some minimal functions:

- **DllMain**: the main entry point when the plugin is loaded, here you execute your code for the 1st time.
- **DllGetClassObject**: here you return the interfaces associated with your plugin.
- **DllCanUnloadNow**: if the plugin is still doing something, don't allow the target program to unload it.
- **DllRegisterServer/DllUnregisterServer**: register/unregister the plugin in Windows registry.

then you define the interfaces that are normally associated with a webcam, you can check [graph-studio-next
](https://github.com/cplussharp/graph-studio-next) for that. You use pipes for sending the [frames](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/dshow/VCamIPC/src/ipcbridge.cpp) to the plugin. Alternatively you can use a [Windows service](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/dshow/Assistant/src) that acts as a "router" if you want to share the frames between many target programs, and source instances.
Each DirectShow plugins instance is independent from each other so there is no problem capturing and sharing at same time, obviously you can't capture and share from the virtuial camera at same time or you will get a deadlock.

>I have the webcam video stream with all applications but I get device not found: -5 with the uvc_open() command.

Because this does not create a virtual USB device, libuvc, and in consequence libusb, talk with the device (or virtual device) using the USB protocol, and for creating a virtual USB device you need to program a driver that create both, an USB interface and an AVStream interface.

hipersayanX, 2019-03-05:
Question answered I think, closing.

closed 2019-03-05 by hipersayanX