webcamoid/webcamoid #164: is it possible to control webcamoid fx parameters via midi or osc?

vincenzoml, 2019-03-05:
Hi. I am in love with the effectiveness of webcamoid at applying effects. It's super-fast and easy to use. Can fx parameters be controlled via midi or osc? I'd love to sync video fxs to bmp in live performances with bitwig.

hipersayanX, 2019-03-05:
I'm not familiarized with midi/osc, how exactly would it work?  can you give a list of programs that use that feature.

vincenzoml, 2019-03-09:
Hi, VJ programs use such features. The idea is that effect parameters and controls (basically whatever one thinks is appropriate) can be changed via messages sent to the application, for real-time changes to FX parameters and things like that (sync video fx intensity to music, for instance), and for play/pause/record control

This can happen

1) because the application is connected to a midi port (and hence changes can be made via programs, or via midi controllers, e.g. the knobs on a midi keyboard); normally controls have a "midi learn" feature, user clicks on a textbox with a number in it, and the program goes into a "waiting" mode, then the user touches a control and the program links that midi message to the parameter

or 

2) because the application listens to the OSC protocol (via UDP I believe). OSC carries text messages and numeric parameters; the application has a public list of messages it responds to and their parameters. 

Approach #2 is much easier to implement and requires less work from a programmer's point of view, but requires more work on the side of the musician. 

Depending on how your application is written it could really be trivial to add OSC support, but nevertheless, if it's not there, obviously I won't expect it to happen right now or to happen at all.

YevgenJohn, 2020-05-23:
Perhaps it would be simpler if the application just to listen to a TCP socket for some text commands, like 'recording start' or 'recording stop', while a trivial alsa-midi-dumper and python code on Raspberry Pi will be doing MIDI integration part and talking over TCP to this software. I already use something similar to control my second piano MIDI playback, so it would be awesome I can start/stop recording using the same MIDI foot pedal. Socket implementation would make this software MIDI independent, thus less maintenance on MIDI problematic environment in Windows.