webcamoid/webcamoid #167: Webcamoid - frame rate

Martin1CZ, 2019-03-10:
[image]
Do someone know how to solve/set webcamoid to capture a video with frame rate 30?
Thanks

hipersayanX, 2019-03-11:
It must take the fps from the source, that is if you where recording your webcam or desktop at 3 FPS, it will record at 3FPS, if this not the case it can be considered a bug.

Martin1CZ, 2019-03-11:
Thanks for feedback, I checked the webcamiod and found FPS set to 30. So maybe it'll be a bug. :-)

Martin1CZ, 2019-03-16:
I just found out that when I set a video format as Matroska and a codec as mpeg4 - MPEG-4 part 2, then my captured video is done with FPS 30.000300... so what can be wrong is still secret for me.

hipersayanX, 2019-03-18:
Will check, thanks!

hipersayanX, 2019-09-18:
I have tested it and it seems that it's setting the frame rate at the real frame rate at which the camera is recording, so even if you set the camera at 30 FPS the real FPS could be slower than that, either because the camera is slow capturing or the codec is slow recording. Nothing can be done to fix this issue since the cause is external to Webcamoid.

closed 2019-10-01 by hipersayanX