webcamoid/webcamoid #168: Is there a way to toggle visibility of the UI?

TSedlar, 2019-03-12:
It'd be nice if there was a way to hide the UI bar if it is not already possible. Something as simple as Ctrl+H to toggle it.

It works surprisingly better than most of the other camera viewers I've used for directly viewing a capture card to play a game (no noticeable audio/input delay)

[image]

But would work much better if the UI was toggleable, simply for downscaling for a smaller window:

[image]

stefan-reich, 2019-05-02:
This would be extremely nice. I'm looking for a low-CPU-use webcam preview window for Linux and with this feature, Webcamoid would be perfect.

crimzind, 2019-06-03:
It's something I'd also like.

bryanmtdt, 2020-04-25:
That would be a great feature, this program catch video source better than others.

stefan-reich, 2020-04-26:
Guys, I'm currently simply using mplayer for webcam preview:

`mplayer -title "WebCam" -tv device=/dev/video2:width=640:height=320 tv://`

Maybe that works for you too?

hipersayanX, 2023-03-05:
Closing this and joining with #581.

closed 2023-03-05 by hipersayanX