webcamoid/webcamoid #169: Android Port

hipersayanX, 2019-03-16:
**Stage 1**: Make it work
Ready :heavy_check_mark:

[image]

 :sunglasses:

hipersayanX, 2019-04-24:
**Stage 2.0**: Camera working (both JNI and NDK) :heavy_check_mark: 

[image]

hipersayanX, 2019-04-27:
**Stage 2.1**: Screen capture working :heavy_check_mark:

[image]

hipersayanX, 2019-07-18:
**Stage 2.2**: Video and audio playback working :heavy_check_mark:

[image]

hipersayanX, 2019-08-17:
**Stage 2.3**: Video and audio recording working :heavy_check_mark:

[image]

And now Webcamoid has all base features working in Android, obviously there would be some bugs out there that needs to be fixed, and Webcamoid is not usable at all yet, but over all the most important part of the port is done.
The next step is reworking the UI, and I'm planning a big full rewrite of it, I will keep you updated.

hipersayanX, 2019-09-02:
[Android packages](https://bintray.com/webcamoid/webcamoid/webcamoid/daily#files/android) are ready for testing. The packages are signed with a debug sign so you will need to enable Android to install from unknown sources. You need Android 6.0 or higher.
This is **NOT** an official release, and as said before this is not ready for daily usage, expect a **LOT** of bugs.

blackchineykh, 2019-10-10:
Hi, are these packages available for download? The link doesn't give us a way to download.

Also does this allow override of the actual physical camera on android? This is my use case and would love to see if I could use for this and also lend any support

hipersayanX, 2019-10-10:
@blackchineykh yes in the **Files** section in the **android** folder. Bintray seems to be a bit bugged and it does not show it directly :confused:

hipersayanX, 2022-01-14:
Ok, after 3 years of not updating, and having several problems with Android packages in CI environments, the Webcamoid port is up and running once again.

[image]

The UI looks clunky, and there are some bugs to solve yet, but the most important part is that at least it's working and the port can keep going forward.

hipersayanX, 2022-02-15:
I found that is crashing when making a screen capture, the port will need a bit more of work.

hipersayanX, 2024-10-01:
Webcamoid 9.2.0 is fully working in Android, the next step is uploading to the PlayStore.

closed 2024-10-01 by hipersayanX