webcamoid/webcamoid #172: webcamoid 8.6 fresh installation does not start (bug?)

ronyshtamler, 2019-03-16:
## Steps to Reproduce

After Installation, I run the icon from the desktop, and nothing happens.
Tried running as Admin both the BAT file and the executable.

## Your Environment
Windows 10 x64

hipersayanX, 2019-03-16:
Try [resetting the configs](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#windows), and then run the portable version, does it works?

ronyshtamler, 2019-03-17:
There are no keys in the registry to reset. Maybe 
I should try to add the keys manually?

hipersayanX, 2019-03-18:
>I should try to add the keys manually?

No, just run the **portable** version.

ronyshtamler, 2019-03-18:
I was under the impression that the downloadable installation version of windows is the portable version.

Maybe some clues in the following installation log?

> ************************************* Invoked: Sat Mar 16 11:38:09 2019
[0] Arguments: D:\Downloads\webcamoid-8.6.0-win64.exe
[6] Operations sanity check succeeded.
[14362] Tmp is on a different volume than the installation directory. Tmp volume mount point: "C:\\" Free space available: "259.68 GiB" Install volume mount point: "D:\\" Free space available: "3.20 TiB"
[14362] Installation space required: "168.55 MiB" Temporary space required: "0.00 bytes" Local repository size: "0.00 bytes"
[15266] backup  operation: Mkdir
[15266] 	- arguments: D:/Program Files/webcamoid
[15266] Done
[15266] perform  operation: Mkdir
[15266] 	- arguments: D:/Program Files/webcamoid
[15272] Done
[15273] Install size: 1 components
[15278] backup com.webcamoidprj.webcamoid operation: Extract
[15278] 	- arguments: installer://com.webcamoidprj.webcamoid/8.6.0bin.7z, D:\Program Files\webcamoid
[15278] Done
[15278] perform com.webcamoidprj.webcamoid operation: Extract
[15278] 	- arguments: installer://com.webcamoidprj.webcamoid/8.6.0bin.7z, D:\Program Files\webcamoid
[17391] Done
[17391] backup com.webcamoidprj.webcamoid operation: Extract
[17391] 	- arguments: installer://com.webcamoidprj.webcamoid/8.6.0content.7z, D:\Program Files\webcamoid
[17391] Done
[17391] perform com.webcamoidprj.webcamoid operation: Extract
[17391] 	- arguments: installer://com.webcamoidprj.webcamoid/8.6.0content.7z, D:\Program Files\webcamoid
[17399] Done
[17399] backup com.webcamoidprj.webcamoid operation: Extract
[17399] 	- arguments: installer://com.webcamoidprj.webcamoid/8.6.0lib.7z, D:\Program Files\webcamoid
[17399] Done
[17399] perform com.webcamoidprj.webcamoid operation: Extract
[17399] 	- arguments: installer://com.webcamoidprj.webcamoid/8.6.0lib.7z, D:\Program Files\webcamoid
[21536] Done
[21536] backup com.webcamoidprj.webcamoid operation: Extract
[21536] 	- arguments: installer://com.webcamoidprj.webcamoid/8.6.0share.7z, D:\Program Files\webcamoid
[21536] Done
[21536] perform com.webcamoidprj.webcamoid operation: Extract
[21536] 	- arguments: installer://com.webcamoidprj.webcamoid/8.6.0share.7z, D:\Program Files\webcamoid
[21704] Done
[21704] backup com.webcamoidprj.webcamoid operation: CreateShortcut
[21704] 	- arguments: D:\Program Files\webcamoid/bin/webcamoid.exe, D:\Program Files\webcamoid/webcamoid.lnk
[21705] Done
[21705] perform com.webcamoidprj.webcamoid operation: CreateShortcut
[21705] 	- arguments: D:\Program Files\webcamoid/bin/webcamoid.exe, D:\Program Files\webcamoid/webcamoid.lnk
[21808] Done
[21808] backup com.webcamoidprj.webcamoid operation: CreateShortcut
[21808] 	- arguments: D:\Program Files\webcamoid/bin/webcamoid.exe, C:\Users\rony\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Webcamoid/webcamoid.lnk
[21808] Done
[21808] perform com.webcamoidprj.webcamoid operation: CreateShortcut
[21808] 	- arguments: D:\Program Files\webcamoid/bin/webcamoid.exe, C:\Users\rony\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Webcamoid/webcamoid.lnk
[21850] Done
[21850] backup com.webcamoidprj.webcamoid operation: CreateShortcut
[21850] 	- arguments: D:\Program Files\webcamoid/bin/webcamoid.exe, C:\Users\rony\Desktop/webcamoid.lnk
[21850] Done
[21850] perform com.webcamoidprj.webcamoid operation: CreateShortcut
[21850] 	- arguments: D:\Program Files\webcamoid/bin/webcamoid.exe, C:\Users\rony\Desktop/webcamoid.lnk
[21887] Done
[21887] backup com.webcamoidprj.webcamoid operation: License
[21887] 	- arguments:
[21887] Done
[21887] perform com.webcamoidprj.webcamoid operation: License
[21887] 	- arguments:
[21892] Done
[21897] Writing maintenance tool: "D:\\Program Files\\webcamoid/WebcamoidMaintenanceTool.exe.new"
[22018] Wrote permissions for maintenance tool.
[22409] Maintenance tool restart: false.
[44159] Warning: QString::arg: Argument missing: Launch Webcamoid now!, Webcamoid
[47077] starting "D:\\Program Files\\webcamoid/bin/webcamoid.exe" ()

ronyshtamler, 2019-03-18:
excuse my mistake. I just found out that there are actually two downloadable links (one for installation and one for portable).
The portable version **works!** - Thank you for your assistance.

hipersayanX, 2019-03-19:
In the rush of the latest release, I manually packaged the installer and in the build I have included the debug symbols, that's why is not launching properly. I will try to automate that part of the build with AppVeyor.

hipersayanX, 2019-03-26:
Fixed in 8.6.1.

closed 2019-03-26 by hipersayanX

DexyZL2, 2019-05-08:
Install in Kubuntu 64-bit 18.04 fully updated - by package 8.1.0+dfsg-7 from repository, or in Windows 10 by webcamoid-installer-8.6.1-x86_64.run script or webcamoid-8.6.1-win64.exe, (on hp Pavilion p6) results in half-hearted icon  that does nothing. Cheese works, but lacks controls/facilities - I'm hoping for something more usable.