webcamoid/webcamoid #174: Issue with code signing virtual camera plugin in Mac OSX 10.14

NODESPLIT, 2019-04-04:
Hey guys, first off really love this project!

I'm currently trying to use your virtual camera plugin for a project on MacOS Mojave (10.14) and having a lot of issues getting it to show up as a device. I can create and manage the virtual camera in webcamoid but it doesn't show up anywhere else.

Taking a look with ```system_profiler SPCameraDataType``` gives me this error:
```
Error loading /Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera:  dlopen(/Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera, 0x0109): code signature in (/Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera) not valid for use in process: mapped file has no cdhash, completely unsigned? Code has to be at least ad-hoc signed.
Cannot find function pointer akPluginMain for factory 41764B79-7320-5643-616D-363462697473 in CFBundle/CFPlugIn 0x7fc7f3e46260 </Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin> (framework, not loaded)
```

And when I attempt to codesign it it throws this error:
```
Error loading /Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera:  dlopen(/Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera, 0x0109): code signature in (/Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera) not valid for use in process: mapping process is a platform binary, but mapped file is not
Cannot find function pointer akPluginMain for factory 41764B79-7320-5643-616D-363462697473 in CFBundle/CFPlugIn 0x7fb3c46159f0 </Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin> (framework, not loaded)
```

Apologies if this is just due to my inexperience with manual code signing and plugin architecture.

hipersayanX, 2019-04-05:
I can't test code sign right now, but making some quick search some people suggest that this is a dead end ([1](https://github.com/inket/cosyTabs/issues/15) [2](https://github.com/sickill/stderred/issues/51)), and some other people suggest this could be solved disabling [SIP](https://www.imore.com/how-turn-system-integrity-protection-macos) ([1](https://github.com/lionheart/openradar-mirror/issues/20512) [2](https://github.com/neonichu/extending-xcode-8/blob/master/nsspain.md#xcode-8-is-under-sip) [3](https://github.com/lionheart/openradar-mirror/issues/20651) [4](https://github.com/kuglee/BatteryExtra/issues/2)).
Also, in programs that does not enforce signing like Firefox and Chrome/ium it may work fine, at least based on the tests I did in the past.

NODESPLIT, 2019-04-05:
Ok thanks for the input @hipersayanX, much appreciated. I came to the same conclusions :(

hipersayanX, 2019-04-06:
Just in case, did you used an ad-hoc code sign ([1](https://apple.stackexchange.com/questions/288291/what-are-the-restrictions-of-ad-hoc-code-signing) [2](https://developer.apple.com/documentation/security/seccodesignatureflags/kseccodesignatureadhoc)) or an official one? because that could make some difference. Also, what target program did you tried?

closed 2019-10-01 by hipersayanX