webcamoid/webcamoid #175: Webcamoid on Windows 7 x64

jsa2340, 2019-04-11:
## Summary

Upon executing Webcamoid a window opens with a white background.
After a short time (Not Responding) is appended to the name in the window title bar.
Under processes in task manager CPU usage is shown as 00
Webcamoid does not load beyond the window with white background.

## Current Behavior
GUI is not displayed in the window

## Expected Behavior
Window is filled with a GUI

## Steps to Reproduce
Open the installed version by clicking the icon
or
Open the portable version by executing the bat 

## Suggestions and tips
Nil

## Your Environment

* Webcamoid information: 
8.6.1-win64
or
portable-daily-win32
or
portable-daily-win64

* Operating System information: Windows 7 Ult x64 SP1

hipersayanX, 2019-09-08:
This bug is probably because D3DCompiler_43.dll and D3DCompiler_47.dll files are missing in webcamoid.exe folder. Anyway, Windows 7 is reaching it's EOL so I won't pay much attention to this bug.

hipersayanX, 2019-10-01:
Windows 7 is approaching its EOL so closing.

closed 2019-10-01 by hipersayanX