webcamoid/webcamoid #176: Webcamoid won't start KDE Neon

rstreeter, 2019-04-11:
After running the Run installer I clicked launch now and the app icon showed up then quickly disappeared then I tried launching it from the app menu 
got the same results. I then opened a terminal and dropped the binary in and got this: qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: linuxfb, minimal, offscreen, vnc, xcb.

Aborted

I did the same with the appimage and got the same results.

I have tried the appimage version as well and the same results.

 Desktop Envrionment : KDE Plasma

*** Webcamoid information:** Version 8.6.1 (x86_64) The Run insataller
*** Operating System information:** KDE Neon 5.15 x86_64

hipersayanX, 2019-04-11:
What distribution?

rstreeter, 2019-04-12:
KDE Neon is the distro it is based on Ubuntu 18.04

asmerovski, 2019-04-20:
I also have the same issue on KDE Neon.
DISTRIB_ID=neon
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="KDE neon User Edition 5.15"

eyalsh, 2019-06-02:
Similar issue on Ubuntu 16.04

Installed by running webcamoid-installer-8.6.1-x86_64.run

eyalsh@arthur:~/webcamoid$ ./webcamoid.sh
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: linuxfb, minimal, offscreen, vnc, xcb.

Aborted (core dumped)

hipersayanX, 2019-09-08:
Tested with Webcamoid daily in all 3 flavors of the distro and everything working well.

closed 2019-09-08 by rstreeter

timwaters, 2020-04-11:
I had to run `sudo apt install libxcb-xinerama0` on Ubuntu 16.04