webcamoid/webcamoid #177: Maximizing window doesn't work correctly with dual-monitor setup

jzarl, 2019-04-11:
## Summary

When I maximize the webcamoid window, instead of filling up the current screen it fills up half of both attached screens/monitors, i.e. it is centered on the edge between both screens.

## Current Behavior

I also tried arranging my screens in Windows so that screen 1 is above screen 2. In this case, the same problem appears, just vertically.
Possibly related, when I launch the application the window appears halfway between screen 1 and screen 2.

## Expected Behavior

Maximizing the window should work like every other application and maximize the window to fill the current screen (only).

## Steps to Reproduce

1.  Open webcamoid
2.  Maximize the application window (either by clicking the "Maximize Window" icon or by grabbing the window title and pushing it against the upper screen border)
3.  Observe that the window not only maximized, but also moved and is now centered between both monitors.

## Your Environment

* Webcamoid information: 8.6.1 Windows 64bit installable
* Operating System information: Windows 10
* Any other useful information: Dual-screen

MikhailRyazanov, 2020-01-02:
I can add that (in dual-monitor configuration under Win10) resizing the window by dragging its edge or corner leads to some insane behavior: the window jumps around and resizes erratically, even if initially it was completely within one monitor area. This is a serious usability issue, since it makes proper resizing almost impossible.

hbh7, 2020-03-23:
Just noticed this too. It's completely unusable for me unfortunately.

hipersayanX, 2020-03-25:
Just in case, this could be a Qt bug since Webcamoid doesn't do any special control on the windowing system. Can someone check if there a similar issue in [Qt bug tracker](https://bugreports.qt.io/secure/Dashboard.jspa)?

MikhailRyazanov, 2020-03-25:
I have no experience with Qt, but the following code
https://github.com/webcamoid/webcamoid/blob/18ac47c4239255a6d13a3bcd5d94d1805b2ad491/StandAlone/share/qml/main.qml#L36-L39
looks suspicious, since the description of these `Screen.desktopAvailable...` [is](https://doc.qt.io/qt-5/qml-qtquick-window-screen.html#desktopAvailableHeight-attached-prop):

> This contains the available height/width of the collection of screens which make up the virtual desktop...

I don't think that positioning with respect to “the collection of screens” is very meaningful, but this resembles very much the currently observed behavior.

hipersayanX, 2020-03-25:
@MikhailRyazanov thanks! did you tried commenting those lines and running?
Also, I still want to position the window in the center on startup,  what would be a good alternative? Screen.width and Screen.height?

MikhailRyazanov, 2020-03-25:
As I said, I have no experience with Qt (and do not have it installed to try recompiling), so unfortunately I cannot help much in this respect...

There might be some relevant information at [QML - main window position on start (screen center)](https://stackoverflow.com/questions/8970766/qml-main-window-position-on-start-screen-center), starting from the second answer.

commit 3f1d6ff2 (2020-03-26): Removed old FFmpeg code. Using Screen.width and Screen.height instead of
Screen.desktopAvailableWidth and Screen.desktopAvailableHeight, this may
fix #177. Removed -q option in Windows launcher, this may fix #226.
closed 2020-03-26 by hipersayanX