webcamoid/webcamoid #178: Cameras still is option in skype after uninstall webcamoid

OsirisDM27, 2019-04-17:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

Write here a brief description of the problem.

## Current Behavior

Describe the problem the best as you can, don't omit information.

## Expected Behavior

How it should have work?

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: (version) (architecture)
* Operating System information: (name) (numeric version and codename if applied) (architecture)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questinons and other matters?

Go ahead!

hipersayanX, 2019-04-19:
Installer does not handle virtual camera uninstall yet, you must remove all virtual camera first and then uninstall Webcamoid.

OscarSilve123, 2019-11-30:
This happens on Linux too.   Would be nice if the maintenance tool/uninstaller prompted to remove all virtual webcams, as well as clean up all the hidden files it leaves around like in .cache and .config on Linux

pedestriantactics, 2020-03-29:
I had an issue where the app began crashing on launch. Now it crashes even if I reinstall. How can I remove the camera without launching the app??

hipersayanX, 2020-03-30:
@OscarSilve123 Linux is a different thing, the virtual camera is managed by your package administrator, Webcamoid can't uninstall the driver.

@pedestriantactics create a _.bat_ file with the following content:

```
@echo off
chcp 1252
regsvr32 /s /u "C:\Program Files\AkVirtualCamera.plugin\x64\AkVirtualCamera.dll"
"C:\Program Files\AkVirtualCamera.plugin\x64\AkVCamAssistant.exe" --uninstall
reg delete "HKLM\SOFTWARE\Webcamoid\VirtualCamera\Cameras" /f
rmdir /s /q "C:\Program Files\AkVirtualCamera.plugin"
```

and execute as administrator.

pedestriantactics, 2020-04-01:
@hipersayinX any clue how I’d do this on a Mac?

hipersayanX, 2020-04-04:
@pedestriantactics

```
launchctl unload -w ~/Library/LaunchAgents/org.webcamoid.cmio.AkVCam.Assistant.plist
rm -f ~/Library/LaunchAgents/org.webcamoid.cmio.AkVCam.Assistant.plist
rm -f ~/Library/Preferences/org.webcamoid.cmio.AkVCam.Assistant.plist
sudo rm -rf /Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin
```

FritzUnterbrink, 2020-04-08:
@hipersayanX after following your steps using the .bat file, I went to discord to use their camera preview to check if it worked. What changed is that when a virtual camera is previewed, the output will be black instead of the standard webcamoid screen. The problem is that all virtual cameras still show up as options for my camera which clutters it up. Restarting the pc did not work. Do you know how the virtual cameras can be deleted from the camera selection list?

hipersayanX, 2020-04-12:
@FritzUnterbrink I've updated the comments, close all programs and run the script again.

FritzUnterbrink, 2020-04-15:
@hipersayanX Nothing changed, but I think I closed all programs that were using my camera. I even disabled all the camera settings in windows and restarted. Could this be fixed in another way?

smileymalaise, 2020-04-15:
I have the exact same program. This program cannot create a good picture in my webcam so I uninstalled it and now I have a phantom virtual webcam in every program and it cannot be deleted.

This software is now categorized as malware. Can you please come up with a valid way to actually remove the garbage your app has created on my computer? thanks.

hipersayanX, 2020-04-16:
@FritzUnterbrink Try with [GraphStudioNext](https://github.com/cplussharp/graph-studio-next), Go to _Graph > Insert filter > Video Capture Sources > Select the phantom camera > Unregister_.

@smileymalaise Since you are a malware expert, can you tell me in which category does it falls? :clown_face: 

https://en.wikipedia.org/wiki/Malware
https://www.avast.com/c-malware
https://www.csoonline.com/article/2615925/security-your-quick-guide-to-malware-types.html

FritzUnterbrink, 2020-04-16:
@hipersayanX Going to video capture sources is where I got stuck. It does not show up for me in the list. Either that or I am in the wrong list. Do you have any advice?

Picture of what I see:
[image]

hipersayanX, 2020-04-16:
@FritzUnterbrink Open ```regedit``` and try finding the phantom camera name, it must appear somewhere in this two places:

```
Computer\HKEY_CLASSES_ROOT\CLSID\{860BB310-5D01-11d0-BD3B-00A0C911CE86}\Instance\{D550C587-BB18-172F-9E8B-4291AD2A40FD}\FriendlyName
Computer\HKEY_CLASSES_ROOT\CLSID\{D550C587-BB18-172F-9E8B-4291AD2A40FD}
```

**Don't do anything**, just tell me if it appears in a similar path.

FritzUnterbrink, 2020-04-17:
@hipersayanX CLSID does not show up for me.
This is as far as I could follow the path: 
[image]

hipersayanX, 2020-04-17:
@FritzUnterbrink but the camera name does not appears anywhere in the registry? what about finding **AkVirtualCamera** or **AkVirtualCamera.dll**?

FritzUnterbrink, 2020-04-17:
@hipersayanX These are the results after searching for AkVirtualCamera
[image]
[image]
[image]
[image]
[image]

And this for AkVirtualCamera.dll
[image]

hipersayanX, 2020-04-17:
@FritzUnterbrink Yes, that was I was looking for.  
For example, in the **Your first path:** capture you can delete the {D550C587-...} key, and all keys containing the **/akvcam/video** **DevicePath**,  

**DON'T delete the top key ({860BB310-...}) or any key not containing such string**.    

Try if that removes the phantom camera.

FritzUnterbrink, 2020-04-18:
@hipersayanX Done and problems solved! No more phantom cameras!
What I did was go to this location and delete all folders inside of "Instance". All folders inside of "Instance" contained a file named what I had named my virtualcams when I made them. After deleting these folders I went to discord to check and no phantom cameras showed up!

Thank you for your help. I appreciate it a lot :)

hipersayanX, 2020-04-19:
Glad to see it worked :smile: 
Usually, that happens because the users try to uninstall the cameras while still using it, and the references are not properly deleted. It will be fixed for the next release for sure.

itsmeboy0, 2022-01-15:
> @OscarSilve123 Linux is a different thing, the virtual camera is managed by your package manager, Webcamoid can't uninstall the driver.
> 
> @pedestriantactics create a _.bat_ file with the following content:
> 
> **NOTE**: Ensure that all programs using the webcam are fully closed (ensure they are not running in background) before running the script. Right click close all the programs, or kill them.
> 
> ```
> @echo off
> chcp 1252
> regsvr32 /s /u "C:\Program Files\AkVirtualCamera.plugin\x64\AkVirtualCamera.dll"
> regsvr32 /s /u "C:\Program Files\AkVirtualCamera.plugin\x86\AkVirtualCamera.dll"
> "C:\Program Files\AkVirtualCamera.plugin\x64\AkVCamAssistant.exe" --uninstall
> "C:\Program Files\AkVirtualCamera.plugin\x86\AkVCamAssistant.exe" --uninstall
> reg delete "HKLM\SOFTWARE\Webcamoid\VirtualCamera\Cameras" /f
> rmdir /s /q "C:\Program Files\AkVirtualCamera.plugin"
> ```
> 
> and execute as administrator.

Thanks, I was going crazy over NVIDIA Broadcast not detecting any of my camera sources until I found out that Webcamoid wasn't properly uninstalled.

hipersayanX, 2022-01-18:
This bug is already fixed in the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), and the changes will be released with Webcamoid 9 (soon).
I decided to split Webcamoid and the virtual camera as different packages, so it will be possible to uninstall the virtual camera after uninstalling Webamoid, even if the user forget to uninstall the virtual camera before.

hipersayanX, 2022-02-15:
This problem was solved in 9.0.0, the virtual camera comes with its own installer.

closed 2022-02-15 by hipersayanX