webcamoid/webcamoid #179: Virtual Webcam didn't uninstall - Can't open program after reinstalling

Nshabdue, 2019-04-20:
I installed 2 virtual cameras with the same name accidentally. While I got an error doing this within webcamoid, it did however make 2 identical webcams visible when I go somewhere like camtests.com. I uninstalled the application, but the virtual cameras remain. They are not in device manager (this is a windows 32-bit install), so I have no clue how to manually remove them. 

I tried to reinstall the application, but it won't open at all. I've uninstalled an reinstalled 2x. I suspect my mistake of reusing the same name on a 2nd camera is causing an issue somehow. 

Is there a way to manually remove the virtual webcam?

Nshabdue, 2019-04-20:
Disregard. I was able to fix this issue by using the portable version of daily build. I removed the devices there.

hipersayanX, 2019-04-21:
As reported in #178, the uninstaller does not uninstall the driver yet, but it will do in future releases. Also there is no problem using the same name for the cameras since each camera is assigned with a different Device ID.

Nshabdue, 2019-04-21:
Gotcha. I searched and couldn't find a similar situation. Thanks for a great product. I'm highly looking forward to v10 with picture in picture!

GetzMikalsen, 2020-02-05:
Experienced the same issue on the latest build for Mac.
Using the portable version and disabling the devices worked fine.

lucario, 2020-03-23:
How do you remove it manually, neither portable or installed version opens at all.

h3llb3nt, 2020-03-24:
same problem as OP but portable app won't run at all. tried re-installing new regular app but it also will not run and crashes on start. have a "test" camera that i cannot get rid of in other video processing apps like OBS.

smileymalaise, 2020-04-15:
I've deleted all files and folders relating to webcamoid and virtualcamera, i've deleted all registry keys related to those as well yet the virtual camera still shows up in discord and other apps. This app is malware.

hipersayanX, 2020-04-16:
@h3llb3nt Try with [this](https://github.com/webcamoid/webcamoid/issues/178#issuecomment-614395023).

@smileymalaise :clown_face: 

https://en.wikipedia.org/wiki/Malware
https://www.avast.com/c-malware
https://www.csoonline.com/article/2615925/security-your-quick-guide-to-malware-types.html

Leemusa, 2021-12-24:
This has nothing to do with Malware.

The issue / solution outlined by   j256   in https://github.com/webcamoid/webcamoid/issues/144 explains the entire situation. j256 offers a terminal solution, but the user can just visit the directories he identifies and delete the files he notes and all is good.

Who ever closed bug https://github.com/webcamoid/webcamoid/issues/144 screwed up big time. Why close a bug before it is no longer a bug?

hipersayanX, 2021-12-27:
@Leemusa

>Who ever closed bug #144 screwed up big time. Why close a bug before it is no longer a bug?

Because the user was asking how to uninstall Webcamoid in GNU/Linux, the user did not give any more interaction, question answered, issue closed.

hipersayanX, 2022-02-15:
This problem was solved in 9.0.0, the virtual camera comes with its own installer.

closed 2022-02-15 by hipersayanX