webcamoid/webcamoid #181: Webcamoid isn't working at all (windows 10

igdegoo, 2019-05-03:
* Webcamoid information: 8.6.1 x64
* Operating System information: Windows 10 x64
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.) dunno where they are, if you need any tell me where to find them

webcamoid plays broken audio of a video i opened in a previous version a year ago, then crashes

i cant really explain anything else in words so have a video

https://youtu.be/ETjo95MdkVg

hipersayanX, 2019-05-04:
Is like it play it play a few seconds of the audio and then die, but it never show any video. Does that happens with other videos? Can you upload that video clip?

igdegoo, 2019-05-04:
>Does that happens with other videos?

I can't switch to another video because webcamoid crashes instantly

hipersayanX, 2019-05-07:
Just in case, if Webcamoid crash and don't let you do anything at the start you can disable the auto play in **regedit**, change the key **HKEY_CURRENT_USER\Software\Webcamoid\Webcamoid\StreamConfigs\playOnStart** key to **false**, so the next time you open Webcamoid you can select any media you want (webcam, desktop, video, etc.), and press the play button to test which one is failing.

inthegarage, 2019-06-13:
I think this is related to the most recent Windows 10 update, the error I see when I get called back on handle error is:

[image]

I hope this helps your investigations.

hipersayanX, 2019-06-14:
@inthegarage where do you get that log?

inthegarage, 2019-06-14:
@hipersayanX This was from the code that I use to connect to the video component (so my code).  This is the only error I have and you don't get much from the lower layers unfortunately. I know it's a hackneyed phrase, but this code hasn't changed my side, yet webcamoid is now failing. I'm strongly suspicious of the new Windows 10 offering we've all taken that seems to changed a number of things.
If I swap to the normal web-cam then everything is OK.

Let me know what further assistance you need.

inthegarage, 2019-06-14:
@hipersayanX This appears to be something to do with having webcamoid and the actual camera active? (well my situation anyway), do you know if this is supported?

igdegoo, 2019-06-14:
Ok so i tried loading up other videos (using the regedit thing hiper suggested) and some of them play at super high speeds, scratchy audio or skip like 3 minutes causing desyncs

i havent found a single video file on my pc that isnt messed up

hipersayanX, 2019-06-14:
@hipersayanX oh get it, you are trying to play the virtual camera in a browser, right? that's a completely different issue from the original topic, it's much better if you open an issue with the browser information.

@igdegoo the crackling audio (when audio is ok but cutted) is a known problem in many platforms, and it's due to audio syncing, since Webcamoid is not a video player I did not pay much attention to this because I had other priorities, but it can be fixed, will check for sure.
The other problem could be that the system audio API is returning a wrong sample format, in which case you will hear a noise or the program will crash.
Anyway, it's always useful to have the video samples that are failing so I can have a better idea of how to fix the problem.

igdegoo, 2019-06-15:
> Anyway, it's always useful to have the video samples that are failing so I can have a better idea of how to fix the problem.
The videos i tried were just random videos i downloaded with youtube-dl, dont think i need to upload them since none of the videos in my folder worked 100% (i have 14 rn)

hipersayanX, 2019-06-15:
Ok, I will check then.

davidacampbell71, 2020-04-25:
I have the same issue on Win10.  Webcamoid 8.7.1 crashes when adding an MP4 source.  Then once it has crashed, it won't restart successfully.  I've found that a workaround to this problem is to rename the source file I added so that Webcamoid can't open it any more and then Webcamoid can start successfully without crashing.  Seems that there's an issue where Webcamoid crashes on opening the file.

davidacampbell71, 2020-04-25:
I am happy to upload the video somewhere if it would help.

hipersayanX, 2020-04-27:
@davidacampbell71 [the last time](https://github.com/webcamoid/webcamoid/issues/216#issue-550962427) I wanted to upload a video, I used https://streamable.com, no sign up required.

davidacampbell71, 2020-04-27:
Hey there, download this file https://streamable.com/hq5hbi to reproduce the problem.  As soon as you add it, within seconds, webcamoid dies and won't restart again properly unless you rename the file.

hipersayanX, 2020-04-27:
@davidacampbell71 thanks, will check.

davidacampbell71, 2020-05-09:
able to reproduce the issue?

hipersayanX, 2020-05-11:
@davidacampbell71 I hear a small noise distortion in the right side of the headphones it could be some problem related to the align of the audio samples, will check later.

davidacampbell71, 2020-05-11:
For me, WebCamoid bombs out completely with this video.  Please see https://streamable.com/g8mhmh where I show you what happens.

In the above video, all I am doing is adding the video downloaded from https://streamable.com/hq5hbi as a video source, and webcamoid seems to consistently bomb every time.

Not only that if you restart webcamoid, it immediately bombs again, so it can't start until I rename the file such that webcamoid can't find it and can start.

davidacampbell71, 2020-05-13:
When webcamoid bombs with the above video, this is seen in the Windows event viewer Application logs:

`
Faulting application name: webcamoid.exe, version: 8.7.1.0, time stamp: 0x00000000
Faulting module name: msvcrt.dll, version: 7.0.18362.1, time stamp: 0xf5bdefd7
Exception code: 0xc0000005
Fault offset: 0x00000000000749ec
Faulting process id: 0x848
Faulting application start time: 0x01d629789e47c51b
Faulting application path: C:\Program Files\webcamoid\bin\webcamoid.exe
Faulting module path: C:\WINDOWS\System32\msvcrt.dll
Report Id: 319e5afb-fce8-443d-909d-02646ca5e717
Faulting package full name: 
Faulting package-relative application ID: 
`

hipersayanX, 2020-05-17:
>#1 0x000000006cfcbe58 in qt_plugin_instance ()
>from /cygdrive/c/Program Files/webcamoid/lib/avkys/VirtualCamera.dll
>#2 0x0000000062c4ea8b in avkys!_ZN9AkElement7iStreamERK8AkPacket ()
>from /cygdrive/c/Program Files/webcamoid/bin/avkys.dll

Did you tried disabling the virtual camera?

davidacampbell71, 2020-05-17:
No, all I did was:
1.  Download video from https://streamable.com/hq5hbi
2.  Go to "Configure Sources" in Webcamoid
3.  Click the green Plus (+) sign
4.  Click Search in the dialog that appears
5.  Choose the file downloaded from https://streamable.com/hq5hbi
6.  Every time I do this, Webcamoid bombs out with the above problem

hipersayanX, 2020-05-17:
@davidacampbell71 from the above log, it's marking the virtual camera as the last element to fail. Try removing the virtual camera and disabling the checkbox, and try again.

davidacampbell71, 2020-05-17:
With the virtual camera disabled (not removed), the video from the file plays on screen successfully.

With the virtual camera removed, the video from the file plays on screen successfully.

With the virtual camera added back after removal, Webcamoid bombs again when playing.

hipersayanX, 2020-05-17:
@davidacampbell71 then the bug is on the virtual camera, probably related to this #264.

davidacampbell71, 2020-05-30:
Is there something I can do to the video such as converting it to a different format using ffmpeg, such that this problem won't occur?  I tried some conversions and they didn't achieve anything, the problem still happened.

hipersayanX, 2022-02-15:
Video playback should be working fine in 9.0.0.

closed 2022-02-15 by hipersayanX