webcamoid/webcamoid #182: Try several filters, ends up crashing the app

usercase, 2019-06-04:
## Summary
I try several filters then the app crashes

## Current Behavior
When I try several filters one by one (almost the whole list), then the app crashes instantly.
I dont even get a Apple report to submit.

## Expected Behavior
No crash

## Steps to Reproduce
1. Open the app
2. Configure sources
3. Set Resolution to 1270x720 (max webcam resolution of iMac 5K 2017)
4. Go to filters when you get a webcam stream on your screen
5. Try every filters one by one from top to bottom until it crashes (after around 30/50 filters trial)

## Suggestions and tips
Possibly a memory management problem. The problem didn't appear in low resolution but only in 1270x720.

How would you solve the problem?
Look if there is no memory problem. If the memory is the problem, flush the filter cache memory at every change ?

## Your Environment
- iMac 5K 2017, macOS 10.14.5
- webcamoid 8.6.1, x86_64
**From Console on macOS**

## Logs 
From macOS Console Center

```
erreur	21:27:53.726174 +0200	webcamoid	CMIOHardware.cpp:339:CMIOObjectGetPropertyData the System is exiting

...

erreur	21:27:53.727143 +0200	webcamoid	CMIO_DALA_System.cpp:264:GetPropertyData error 1970171760 (<private>) getting property selector (<private>) scope (<private>) element 0

...

erreur	21:27:54.726252 +0200	webcamoid	HALC_ShellObject.cpp:212:HasProperty:  HALC_ShellObject::HasProperty: initial inner method call failed, Error: 560947818 (!obj)

...

erreur	21:27:56.648609 +0200	webcamoid	        AEGetDescData of non-desc type 'reco' not a good idea
erreur	21:27:58.638507 +0200	webcamoid	CMIO_Unit_ScopeElement.h:200:SafeGetElement Throwing err: -67454
erreur	21:27:58.639663 +0200	webcamoid	CMIOUnitFigBaseObjectImpl.c:246:CMIOUnitCreateFromDescription Invalid paramater
erreur	21:27:58.654180 +0200	webcamoid	CMIO_Unit_Input_Device.cpp:244:GetPropertyInfo CMIOUInputFromProcs::GetPropertyInfo() failed for id 102, Error: -67456

...

error	21:28:09.517945 +0200	webcamoid	CMIO_Unit_ScopeElement.h:200:SafeGetElement Throwing err: -67454
error	21:28:09.518987 +0200	webcamoid	CMIOUnitFigBaseObjectImpl.c:246:CMIOUnitCreateFromDescription Invalid paramater
error	21:28:09.520654 +0200	webcamoid	CMIO_Unit_Input_Device.cpp:244:GetPropertyInfo CMIOUInputFromProcs::GetPropertyInfo() failed for id 102, Error: -67456
```
----------

Hope all those details will help you.

**Thank you for your hard work !**

hipersayanX, 2019-10-01:
Fixed in 8.7.0.

closed 2019-10-01 by hipersayanX

usercase, 2019-10-01:
Seems much more stable with the 8.7.0 

Tried the app a bit and no problem, big thanks 👍