webcamoid/webcamoid #183: Context for translatable strings

comradekingu, 2019-07-14:
https://hosted.weblate.org/zen/webcamoid/main/nb_NO/?type=nottranslated

Could all these strings be explained? I feel annotating these strings would result in higher quality translations. It is also possible for admins to add screenshots.

hipersayanX, 2019-07-15:
I have added comments for the most technical words at least. Adding screenshots for translators is kinda impossible since Weblate does not support such feature afaik, also not sure if it would be possible to automate UI screenshots in an easy way.

hipersayanX, 2019-07-18:
I think the issue is solved so closing. Let me know, If there are more text that needs comments.

comradekingu, 2019-09-30:
@hipersayanX Having the rest of the strings in https://hosted.weblate.org/zen/webcamoid/main/nb_NO/?type=nottranslated commented would really be nice. For example, is equalization to do with sound, or video? The difference could mean the wrong word ends up being used. Only 4 languages are completed in any sense, so I think it would help a lot.

hipersayanX, 2019-09-30:
I have added the comment for Equalize. If there are other words that needs some explanation just tell me.

>Having the rest of the strings in https://hosted.weblate.org/zen/webcamoid/main/nb_NO/?type=nottranslated commented would really be nice.

The problem of commenting everything is that it's a very exhausting work, and there are words or phrases that have a literal translation and for me does not make much sense commenting (for example [all these words](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/FaceDetect/share/qml/main.qml#L334)), and there are some words or phrases that I can understand but other people maybe can't but I'm not aware of it (maybe because the word is too technical but I'm used to it).
That's why for me is much easier if people ask me for comments.

KovalevArtem, 2020-05-01:
@hipersayanX Please update Daily Build
I want to check my translation in the application itself...

hipersayanX, 2020-05-01:
@KovalevArtem Bintray and Github are giving me a lot of problems uploading the daily builds lately :confused:, the first because they have a nonsensical 365 days cap for updating versions, and the second because reasons :man_shrugging:
Will try to fix them this week.

KovalevArtem, 2020-05-02:
@hipersayanX I sincerely regret you. But many thanks for the update!

KovalevArtem, 2020-05-02:
[image]
It seems these video formats are not displayed correctly in the program. Also these lines are not in Weblate.
And in Weblate there are no lines for translating buttons

hipersayanX, 2020-05-05:
@KovalevArtem in the first picture, the strings are provided by FFmpeg or GStreamer, those are not part of Webcamoid source code, it will require some hard work to support translations for that. The same happens for the camera controls.
The second one is fixed.
And the last two, those dialogs are provided by Qt I've no control of those strings.

closed 2022-02-15 by hipersayanX